Klipper3d/klipper #2578: [Ender 3, SKR mini 1.2] Pressure Advance question

sedlacekdavid, 2020-03-02:
Hi Guys, first of all, let me thank you for this amazing firmware.

I'm currently battling with PA. Is there anybody successfully using PA on a bowden setup and don't having swelled corners?

I tried to follow the guide to tweak PA factor, but unfortunately I left it running over the night, so I'm not really sure if PA was in charge or not. Because from the photo below it seems that it is completely same from the bottom all the way to the top. I've realised I probably set Print speed to 120 mm/s in CURA, but this applies just for infill and wall speed is half of that by default. 

Should I expect some better results once increase speed to truly 120 mm/s? I have a suspicion that it's just because of my 50 cm long bowden. I'll do some changes, but obviously I can't below 25-30cm.

I sliced the tower, uploaded to my Octoprint setup, executed commands below, and ran the print.

`SET_VELOCITY_LIMIT SQUARE_CORNER_VELOCITY=1 ACCEL=500

TUNING_TOWER COMMAND=SET_PRESSURE_ADVANCE PARAMETER=ADVANCE START=0 FACTOR=.020`

Pics:
[https://imgur.com/a/NeNRIDK](url)

Regards David

StrikeEagleCC, 2020-03-03:
I've been using pressure advance on a bowden setup with great success. If the test is working, I would expect the results to be pretty obvious.

The commands you've put in your comment are correct--copying and pasting them in my terminal works. Is it possible you mis-typed them? 

Also, the tuning tower is a somewhat recent feature. If you're running an older version, [update](https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#how-do-i-upgrade-to-the-latest-software). If you're running an old version, you will get an "UNKNOWN COMMAND" error. If you are running the newest version and get it right, you should see confirmation that the test is starting, both in the terminal, and at the top of the octoprint screen.

[image]

KevinOConnor, 2020-03-03:
@sedlacekdavid, if you can extract the log file from the print, it will show whether or not the printer successfully started the tuning tower.  The logs are backed up for several days (assuming the raspberry pi is not rebooted).

-Kevin

sedlacekdavid, 2020-03-03:
Guys, I probably made a mistake or dont know, but yesterday I tried it again and I must say the results are impressive! Its amazing how much its visible.

I also shortened my bowden from 45 cm to 20 cm- among the others thanks to moving extruder closer to the middle of X axis. 

On the photos you can see some holes, I noriced extruders filament sometimes “slips” , Maybe not enough pressure force, I probably didn’t mount it back correctly, but thats not an issue since I already ordered a double geared one.

Pics https://imgur.com/gallery/TSNfhCD

sedlacekdavid, 2020-03-03:
Btw I've recently moved to your firmware and the first thing I noticed here is how much is everybody here responsive, thank you very much!

sedlacekdavid, 2020-03-03:
Btw I think on the latest photos I also accidentaly left coasting turned on.

sedlacekdavid, 2020-03-05:
Guys, I'm very impressed by Pressure Advance feature. It works 100 % also on slow speed prints. My factor is currently 0.95

The result is perfect....

closed 2020-03-05 by sedlacekdavid

StrikeEagleCC, 2020-03-05:
@sedlacekdavid Glad you got it working. 0.95 seems pretty high, but if it works, it works! In my experience with _my particular printer_, at this level i was much more likely to get some skipped extruder steps at moderate to high print speeds. If this becomes a problem for you in the future, I'd look into eliminating as much of the slop in your bowden tube couplings as possible, and potentially upgrading to a high quality tube (capricorn or similar).

I have an Ender 3 with a 450mm capricorn tube and just a little slop at the extruder coupling. 0.48 is the PA I use for typical PLA.

Happy printing.

sedlacekdavid, 2020-03-06:
@StrikeEagleCC I already have a capricorn tube and I 99% print only PETG. And I managed to shorten the bowden down to ~20 cm. Regarding the couplings, The chinese ones are crap, so I replaced extruder side by a locally sourced one - a contractor shop, which I think is quite good - you can also feel clicking when pressing. 
But still when I see moving the filament back and forth, there is definitely a small play in bowden, like ~ 0.5-1mm, hard to say. I'm not sure if there's something I can do about that, since the play is better than with chinese connectors, but still it's there. 

I have to admin there is no play on the hotend side where is still the old chinese one. It has different construction, it's slightly bigger.

But ok, that's a point to focus on, I'll somehow manage to completely fix it.

sedlacekdavid, 2020-03-06:
Another point - I've also increased **pressure_advance_smooth_time** to 0.2 as mentioned in the sample config as the "optimal maximum" and the impression was a big surprise, Extruder motor is now elegantly build the pressure when printing at any speed, instead of fast chopping with the value of **0.04** . 
I'm not sure about the side effects, but so far it's doing really great job. So my general impression is YES IT FINALLY WORKS!!!