Klipper3d/klipper #2580: Probe Accuracy

boboxx, 2020-03-03:
I just setup a capacitance proximity Probe (LJC18A3-B-Z/BX) on my Ender 3 and I have been having a hard time getting a repetitive first layer. I have tried doing the probe offset (PROBE_CALIBRATE) when the bed was cold, when it was hot. I'm also doing the BED_MESH_CALIBRATE cold/hot. I'm not sure if it's a process or a config problem but I haven't really been able to get a nice first layer repetitively.

Here is what I do:
- home the probe in my safe Z, this centres the extruder on the bed, 5mm off
- head the bed and probe 5-8min
- run PROBE_CALIBRATE and save_config
- home XY, Z
- reheat the bed
- run BED_MESH_CLEAR
- run BED_MESH_CALIBRATE
- save_config

Here are some output of my last calibration:

end: GET_POSITION
Recv: // mcu: stepper_x:-191 stepper_y:-126 stepper_z:765
Recv: // stepper: stepper_x:118.000000 stepper_y:118.000000 stepper_z:4.999500
Recv: // kinematic: X:118.000000 Y:118.000000 Z:4.999500
Recv: // toolhead: X:118.000000 Y:118.000000 Z:5.000000 E:0.000000
Recv: // gcode: X:118.000000 Y:118.000000 Z:5.122301 E:0.000000
Recv: // gcode base: X:0.000000 Y:0.000000 Z:0.000000 E:0.000000
Recv: // gcode homing: X:0.000000 Y:0.000000 Z:0.000000

Send: PROBE_ACCURACY
Recv: // PROBE_ACCURACY at X:118.000 Y:118.000 Z:5.000 (samples=10 retract=2.000 speed=50.0 lift_speed=50.0)
Recv: // probe at 118.000,118.000 is z=1.087000
Recv: // probe at 118.000,118.000 is z=1.097000
Recv: // probe at 118.000,118.000 is z=1.109500
Recv: // probe at 118.000,118.000 is z=1.102000
Recv: // probe at 118.000,118.000 is z=1.089500
Recv: // probe at 118.000,118.000 is z=1.099500
Recv: // probe at 118.000,118.000 is z=1.089500
Recv: // probe at 118.000,118.000 is z=1.094500
Recv: // probe at 118.000,118.000 is z=1.084500
Recv: // probe at 118.000,118.000 is z=1.079500
Recv: // probe accuracy results: maximum 1.109500, minimum 1.079500, range 0.030000, average 1.093250, median 1.092000, standard deviation 0.008533

Move back to sede of bed and rehome Z to center:
G28 X0 Y0
G28 Z0

Send: PROBE_ACCURACY
Recv: // PROBE_ACCURACY at X:118.000 Y:118.000 Z:5.000 (samples=10 retract=2.000 speed=50.0 lift_speed=50.0)
Recv: // probe at 118.000,118.000 is z=1.094500
[...]
Recv: // probe at 118.000,118.000 is z=1.092000
Recv: // probe at 118.000,118.000 is z=1.092000
Recv: // probe at 118.000,118.000 is z=1.092000
Recv: // probe at 118.000,118.000 is z=1.082000
Recv: // probe at 118.000,118.000 is z=1.089500
Recv: // probe at 118.000,118.000 is z=1.087000
Recv: // probe at 118.000,118.000 is z=1.092000
Recv: // probe at 118.000,118.000 is z=1.087000
Recv: // probe at 118.000,118.000 is z=1.094500
Recv: // probe accuracy results: maximum 1.094500, minimum 1.082000, range 0.012500, average 1.090250, median 1.092000, standard deviation 0.003717

The precision seem to be there and is more then what I can do by eye, but some how most of my print start have to be done 3-4 time before I get bed adhesion and even then, they dnse seem to be all equal.

[image]
[klippy.log](https://github.com/KevinOConnor/klipper/files/4283092/klippy.log)

Hywelmartin, 2020-03-04:
https://youtu.be/il9bNWn66BY?t=689

KevinOConnor, 2020-03-21:
I'm going to close this ticket as it looks like the conversation has concluded.  As far as we know, the software works as intended.  It doesn't look like anyone has any particular tips on improving the accuracy of this particular hardware sensor.

-Kevin

closed 2020-03-21 by KevinOConnor