Klipper3d/klipper #2582: Wrong Extruder Direction

edjrwinnt, 2020-03-04:
I've changed the dir_pin from ar28 to !ar28 and the extruder still goes in reverse. 
 
[Klippy.log](https://github.com/KevinOConnor/klipper/files/4285034/Klippy.log)

boboxx, 2020-03-04:
You need to flip the connection wire, I have never done it but there are 2 coils.

Hywelmartin, 2020-03-04:
it doesn't seems like you saved that change
[image]

edjrwinnt, 2020-03-04:
It's is saved. I made other changes and they were there the next time I went to change something else in the file. I even tried it 2 or 3 times toggling back and forth with and without the !, plus I even tried the ^.

I will try the hardwire route but don't understand why this way will not work. 

Thank you both for your help.

Hywelmartin, 2020-03-04:
well it doesn't show in the log..

closed 2020-03-21 by KevinOConnor