Klipper3d/klipper #2584: PT1000 as thermistor on Duet Wifi (probably everywhere) impossible to set up

RicoKirstein, 2020-03-04:
Since the Duet Wifi is supposed to work with PT1000 on the thermistor inputs, I decided to try it out.
#1137 Had all the information I needed, but I would always get a math domain error.
Manually calculating all the values I tried just setting beta.
Since a PT1000 is essentially a PTC I would need a negative beta value of ~-364.779
That also didn't work since you are only allowing positive beta values.
I removed that restraint in the thermistor.py line 90. 
Now I get reasonable temperature values from the sensor since the c3 value was basicly zero anyway.
It was at like 5.something*10^-5

So you should probably change the comparison in setup_coefficients to round c3 and the check <=0.
and also allow negative beta.

Cheers
Rico

[klippy.log](https://github.com/KevinOConnor/klipper/files/4289574/klippy.log)

KevinOConnor, 2020-03-05:
FYI, a PT1000 isn't a thermistor, and using the thermistor math on it will likely result in funky results.  I'd suggest using the adc_temperature module instead (see config/example-extras.cfg for the details).  That module is specifically made for sensors like the PT100 and PT1000.

-Kevin

RicoKirstein, 2020-03-05:
Thanks for the quick answer.
I somehow over read the whole section for resistance adc and only saw the voltage section.
The adc module does work...
I guess this issue can be closed.

Cheers
Rico

closed 2020-03-05 by RicoKirstein