Klipper3d/klipper #2585: using precision piezo for z probe and negative z_offset

simonlee, 2020-03-05:
Hello,

(i3 style cartesian printer)

My setup is such that when the nozzle hits the heatbed, the piezo sensor (probe z endstop) reports triggered as expected. This means that the nozzle ever so slightly pushes against the bed to activate the endstop trigger.

Initially,

my stepper_z section has the following:
```
position_endstop: 0.0
position_min: 0.0
position_max: 200
```

and my probe section contained:
`z_offset: 0.0`

When I try to probe_calibrate I get a final result of a negative z_offset value, which makes sense. Like z_offset: -0.112. However, klipper gives an error when restarting:

> Recv: !! position_endstop in section 'stepper_z' must be between position_min and position_max

Typically the z_offset is positive since most z probes activate before the nozzle touches the bed. In this case, it activates when the nozzle touches and/or goes past slightly the bed surface. What would be the appropriate position_min and endstop_position values for the z stepper? Or am I hitting a bug?

BlackStump, 2020-03-05:
https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L558

simonlee, 2020-03-05:
Great thanks @BlackStump! I got it working as expected with the suggestions you've provided.

closed 2020-03-05 by simonlee