Klipper3d/klipper #2586: TAZ 6 Single Extruder .cfg not working

reinharm1959, 2020-03-05:
Despite trying several times I can't get Klipper to work in Taz 6 with stock single extruder. After homing the printer immediately disconnects Error "min extruder temp below min" or words to that effect. I found a thread that was several years old but nothing I tried worked so I uninstalled everything and am back to straight Octoprint.  Too bad I was really hoping to see if Klipper resulted in any improvements over the stock firmware. You would hope (expect) that the developers would have fixed the printer-lulzbot-taz6-2017.cfg file to actually work. Oh well.

reinharm1959, 2020-03-05:
Well I’ll see if I still have the log file but the issue is still real and renders Klipper completely unusable in my and many others with a Taz 6. I will try it one more time with a Dual Extruder V3 whose .cfg file seems to take the other prerequisite gcode start, wipe, bed-leveling,  and stop commands into account were the printer-lulzbot-taz6-2017.cfg appears to lack any of the required commands after touching the toggle disc.

reinharm1959, 2020-03-05:
Log file attached

reinharm1959, 2020-03-05:
log file attached

[klippy.log](https://github.com/KevinOConnor/klipper/files/4293090/klippy.log)

jakep82, 2020-03-05:
This is a gcode issue.  M109 R175 should be M109 S175.

reinharm1959, 2020-03-05:
Forgive my neebie-ness but I search the cfg file and can not find any instance of M109 R175 or M109 S175 or finally R175. So in which file in Klipper do these command reside?

 

Thanks

jakep82, 2020-03-05:
Its not in any Klipper file. Look in the start gcode of your slicer.

jakep82, 2020-03-05:
Change the R in each line to S.  The R option is not supported in Klipper.

```
M109 S{material_soften_temperature} ; soften filament before homing Z

M109 S{material_wipe_temperature} ; wait for extruder to reach wiping temp

M109 S{material_probe_temperature} ; wait for extruder to reach probe temp

M109 S{material_print_temperature_layer_0} ; wait for extruder to reach printing temp

reinharm1959, 2020-03-05:
So in start code in Cura (lulzbot) the incidences of M109 are as follows:

 

;This G-Code has been generated specifically for the LulzBot TAZ 6 with standard extruder

M109 R{material_soften_temperature} ; soften filament before homing Z

M109 R{material_wipe_temperature} ; wait for extruder to reach wiping temp

M109 R{material_probe_temperature} ; wait for extruder to reach probe temp

M109 R{material_print_temperature_layer_0} ; wait for extruder to reach printing temp

 

So your saying replace the;

 

M109 R{material_soften_temperature} and {_wipe_} and {probe_temperature} variables with M109 S175?

Or just

M109 R{material_probe_temperature} with M109 S175??

 

Once again in the gcode there is no instance of M109 R175

 

Thanks

reinharm1959, 2020-03-06:
Ok did that and that worked and the after the probe wipe the probe missed the first leveling disc by a mm or two.

KevinOConnor, 2020-03-21:
I'm going to close this ticket as it looks like a solution was found for the original issue.

-Kevin

closed 2020-03-21 by KevinOConnor