Klipper3d/klipper #2589: TMC2208 shut down

Paulbsh, 2020-03-05:
Hi folk,
I just experienced a Z stepper shutdown.
I know that 2208's shouldn't be run in standalone mode. 
They're setup for UART use.
The drivers are cooled actively and are cool to the touch,

After the shutdown, I queried the steppers registers:
Recv: // ========== Write-only registers ==========
Recv: // IHOLD_IRUN: 0008190f IHOLD=15 IRUN=25 IHOLDDELAY=8
Recv: // TPWMTHRS:   00000009 TPWMTHRS=9
Recv: // TPOWERDOWN: 00000014 TPOWERDOWN=20
Recv: // ========== Queried registers ==========
Recv: // GCONF:      000001c0 pdn_disable=1 mstep_reg_select=1 multistep_filt=1
Recv: // GSTAT:      00000003 reset=1 drv_err=1(ErrorShutdown!)
Recv: // IFCNT:      0000003c IFCNT=60
Recv: // OTP_READ:   0000000f OTP_FCLKTRIM=15
Recv: // IOIN@TMC220x: 20000341 ENN=1 PDN_UART=1 SEL_A=1(TMC220x) DIR=1 VERSION=0x20
Recv: // FACTORY_CONF: 0000000f FCLKTRIM=15
Recv: // TSTEP:      000fffff TSTEP=1048575
Recv: // MSCNT:      000002e8 MSCNT=744
Recv: // MSCURACT:   01dc010b CUR_A=-245 CUR_B=-36
Recv: // CHOPCONF:   14030053 toff=3 hstrt=5 TBL=2 vsense=1 MRES=4(16usteps) intpol=1
Recv: // DRV_STATUS: c00f0000 CS_ACTUAL=15 stealth=1 stst=1
Recv: // PWMCONF:    c80d0e24 PWM_OFS=36 PWM_GRAD=14 pwm_freq=1 pwm_autoscale=1 pwm_autograd=1 PWM_REG=8 PWM_LIM=12
Recv: // PWM_SCALE:  0000000f PWM_SCALE_SUM=15
Recv: // PWM_AUTO:   000c001f PWM_OFS_AUTO=31 PWM_GRAD_AUTO=12

I noticed the GSTAT: 
GSTAT:      00000003 reset=1 drv_err=1(ErrorShutdown!)

Is there a way to determine why the stepper did shut down?
Thanks.
[klippy.log](https://github.com/KevinOConnor/klipper/files/4292913/klippy.log)

KevinOConnor, 2020-03-05:
I recommend upgrading to the latest code - PR #2565 was committed just yesterday.

To answer your question though - the TMC_DUMP output doesn't seem to indicate the reason for the shutdown.

-Kevin

Paulbsh, 2020-03-06:
Can I be of any help with the analysis of this case? 
I just had a z shutdown again.
Here's the most recent klippy.log:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4298613/klippy.log)

KevinOConnor, 2020-03-06:
It'll take me several days to get another version of the "step+dir+step" filter available for testing.  In the interim, I recommend you set the stepper_z stealthchop_threshold=8, as it ideally should be just slightly above the maximum velocity for the given axis.  That said, your shutdowns may not be due to the "step+dir+step" issue - it could be due to overheating, electrical shorts, or some other issue.

-Kevin

Paulbsh, 2020-03-06:
I set it to 7 yesterday but just noticed it popped back to 50 for whatever unknown reason, maybe my mistake with having the file open via FTP in Notepad++ and having accidentally overwritten the config.
I could almost eliminate overheating as a possible cause as I'm cooling the drivers with an 80mm fan and they're cool to the touch (I touched the heatsink when the drivers shut down). 
I'll keep an eye to this issue and report back if I find something out

KevinOConnor, 2020-03-07:
If you can try PR #2597 on your printer that would be great.  If so, let me know the results (success or failure).

-Kevin

KevinOConnor, 2020-03-21:
Any further updates on this?  FYI, #2597 was committed last week.

-Kevin

scmsystm, 2020-03-23:
hello, i've been experienced randomly shutdown that causing layershift after #2597 with tmc2208 in stealthchop or spreadcycle.
before i updating my klipper, I'm was success 10hours print.
I dont know if it come from #2597 but after I revert to previous commit and run with spreadcycle, my printer working again printing 5 hours job.
last tmc dump after layershift appear was with stealthchop_threshold: 9999
```
Send: dump_tmc stepper=stepper_y
Recv: // ========== Write-only registers ==========
Recv: // IHOLD_IRUN: 00081b0f IHOLD=15 IRUN=27 IHOLDDELAY=8
Recv: // TPWMTHRS:   00000001 TPWMTHRS=1
Recv: // TPOWERDOWN: 00000014 TPOWERDOWN=20
Recv: // ========== Queried registers ==========
Recv: // GCONF:      000001c0 pdn_disable=1 mstep_reg_select=1 multistep_filt=1
Recv: // GSTAT:      00000003 reset=1 drv_err=1(ErrorShutdown!)
Recv: // IFCNT:      00000006 IFCNT=6
Recv: // OTP_READ:   0000000d OTP_FCLKTRIM=13
Recv: // IOIN@TMC220x: 20000150 DIAG=1 PDN_UART=1 SEL_A=1(TMC220x) VERSION=0x20
Recv: // FACTORY_CONF: 0000000d FCLKTRIM=13
Recv: // TSTEP:      00000181 TSTEP=385
Recv: // MSCNT:      00000041 MSCNT=65
Recv: // MSCURACT:   00db0072 CUR_A=114 CUR_B=219
Recv: // CHOPCONF:   14030053 toff=3 hstrt=5 TBL=2 vsense=1 MRES=4(16usteps) intpol=1
Recv: // DRV_STATUS: 401b0020 s2vsb=1(LowSideShort_B!) CS_ACTUAL=27 stealth=1
Recv: // PWMCONF:    c80d0e24 PWM_OFS=36 PWM_GRAD=14 pwm_freq=1 pwm_autoscale=1 pwm_autograd=1 PWM_REG=8 PWM_LIM=12
Recv: // PWM_SCALE:  00000038 PWM_SCALE_SUM=56
Recv: // PWM_AUTO:   000f0025 PWM_OFS_AUTO=37 PWM_GRAD_AUTO=15
```

[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/4368763/klippy.log.zip)

KevinOConnor, 2020-03-23:
@scmsystm - I'm not sure what log you wanted us to look at.  Make sure you are on the latest version of the code (see the FAQ for upgrade instructions).  If you encounter a problem, run DUMP_TMC on the stepper having the problem, then run M112, and then attach that log.

-Kevin

scmsystm, 2020-03-24:
please forget about my previous comment, I install klipper with fresh and latest one and it seem no problem so far.

KevinOConnor, 2020-04-08:
I'm going to close this as the Klipper workaround for tmc2208 stealthchop quirks has been merged.

-Kevin

closed 2020-04-08 by KevinOConnor