Klipper3d/klipper #2590: MCU shutdown timer too close

Paulbsh, 2020-03-05:
I recommend upgrading to the latest code - PR #2565 was committed just yesterday.

To answer your question though - the TMC_DUMP output doesn't seem to indicate the reason for the shutdown.

-Kevin

_Originally posted by @KevinOConnor in https://github.com/KevinOConnor/klipper/issues/2589#issuecomment-595344930_

I did a git pull a few minutes ago (like advised by Kevin) to solve another issue.
I'm getting mcu shutdown: timer too close after homing x and y since then.

KevinOConnor, 2020-03-05:
Looks like a regression.  Until it can be resolved you'll probably want to return to the previous version (`cd ~/klipper ; git checkout 31b2c3ea ; sudo service klipper restart`).

-Kevin

Paulbsh, 2020-03-05:
> Looks like a regression. Until it can be resolved you'll probably want to return to the previous version (`cd ~/klipper ; git checkout 31b2c3ea ; sudo service klipper restart`).
> 
> -Kevin

Thank you Kevin, I'm able to home again.
I'll close this issue for now

Paulbsh, 2020-03-05:
Sorry to re-open this so quickly.
is 31b2c3ea correct?
I get "Move out of range" when printing the first layer 
[klippy.log](https://github.com/KevinOConnor/klipper/files/4295076/klippy.log)

Paulbsh, 2020-03-05:
Sorry, my fault. My bltouch messed something up while creating the mesh.

KevinOConnor, 2020-03-05:
FYI, I reverted that commit (as of commit e872290c4).  You should be able to go back to the master branch (`cd ~/klipper ; git checkout master ; git pull`).  However, the master branch no longer has the tmc2208 "step+dir+step" filter.

-Kevin