Klipper3d/klipper #2593: Taz 6 ABL not hitting level disc

reinharm1959, 2020-03-06:


reinharm1959, 2020-03-06:
So I modified the gcode in Cura and got Klipper to home and wipe but it missed the first leveling metal disc so to avoid damage I had to kill the print by turning the printer off. I have searched with WinSCP and NO klippy.log file was found or generated so it's not attached.

reinharm1959, 2020-03-06:
Sorry don't know why these were not found in the previous search
[klippy.log](https://github.com/KevinOConnor/klipper/files/4295806/klippy.log)

reinharm1959, 2020-03-06:
another one
[klippy.log](https://github.com/KevinOConnor/klipper/files/4295811/klippy.log)

ld3300, 2020-03-06:
It looks like your bed tilt coordinates are off.  On the machine at work the min x and y are about -9mm.  Do a G28 command to home, then use G1 X and Y commands to move the nozzle over the points.  At each point do a M114 and those will be your coordinates to fix in the config file.

reinharm1959, 2020-03-06:
Also the z probe offset in the cfg file is a positive number (1.30) shoudn’t be -1.30 as in the marlin firmware?

jakep82, 2020-03-06:
> Also the z probe offset in the cfg file is a positive number (1.30) shoudn’t be -1.30 as in the marlin firmware?>

No

closed 2020-03-21 by KevinOConnor