Klipper3d/klipper #2596: [FR] safe_z_home start/end gcode

kakou-fr, 2020-03-06:
Hi,

I use a touch'mi that need a contact with a magnet to deploy the pin. in my case, the magnet is at X230, so I use : 
```
[homing_override]
set_position_z: 10
axes: z
gcode:
    G90
    G1 Z20 F300
    G28 X0 Y0
    G1 X230 F3600
    G1 X68 Y85 F3600
    G28 Z0
    G1 Z0 F300
    G1 Z10 F300
```

It's possible to add a start/end gcode to safe_z_home to call a gcode before the home of Z and after like : 

```
[safe_z_home]
home_xy_position: 68,85
speed: 70.0
z_hop: 10.0
z_hop_speed: 5.0
move_to_previous: False
start_gcode: 
     G1 X230 F3600   # deploy pin
end_gocde:
     G1 Z0 F300          # retract pin
     G1 Z10 F300
```

FHeilmann, 2020-03-09:
You can do what you want the following way:

instead of using an endstop, attach your pin to a `[probe]` section, and use the activate and deactivate gcode of that section to deploy and retract your probe. To use this (fake) probe as your z endstop, set `probe:z_virtual_endstop` in your `[stepper]` block.

kakou-fr, 2020-03-09:
yes, but in the docs, the activate/desactivate is run every probe (and we do not use G1 command). and I must, in my case, execute one G1 to deploy the pin, and at the end of the probe or mesh, one G1 to close the pin : 
```
#activate_gcode:
#   A list of G-Code commands to execute prior to each probe attempt.
#   See docs/Command_Templates.md for G-Code format. This may be
#   useful if the probe needs to be activated in some way. Do not
#   issue any commands here that move the toolhead (eg, G1). The
#   default is to not run any special G-Code commands on activation.
#deactivate_gcode:
#   A list of G-Code commands to execute after each probe attempt
#   completes. See docs/Command_Templates.md for G-Code format. Do not
#   issue any commands here that move the toolhead. The default is to
#   not run any special G-Code commands on deactivation.
```

KevinOConnor, 2020-03-09:
There are a couple of possible solutions:
1. Continue to use `[homing_override]`.
2. Override the G28 command (via a `[gcode_macro G28]` with rename_existing set) and issue your additional commands as part of the new G28 macro.

-Kevin

kakou-fr, 2020-03-09:
ok i will continue to use homing_override.

closed 2020-03-09 by kakou-fr