Klipper3d/klipper #2599: Print head moves out of bed range

insidecircuits, 2020-03-07:
When I try to start my print, the axis do the homing then the print wants to start but the head is moving from center to upper right corner of bed trying to exceed the limit. z
Then octoprint shows Move out of range error. The 3d model is very small, it's not the problem.
I attached the gcode sequence also.
[image]

[out of range.txt](https://github.com/KevinOConnor/klipper/files/4302208/out.of.range.txt)

[klippy.log](https://github.com/KevinOConnor/klipper/files/4302162/klippy.log)

insidecircuits, 2020-03-07:
Update now sometimes I get Move exceeds maximum extrusion error.
[image]

ld3300, 2020-03-07:
Hi @insidecircuits 
We should be able to figure this out.. Please include your entire klippy.log.  It doesn't include all the dump information that would help to find the issue.  The first error is because some command tried to move your y axis out of the range set in your config, the second one would be a command that has a larger extrusion amount than expected for the x/y move happening.  Send a new complete klippy log and your g-code file.  Doing M112 when you encounter the error can be really helpful too because it dumps extra information into the log file.

insidecircuits, 2020-03-07:
Thank you for your response, I use cura to slice the model.
I have set up the machine limits the same in cura, octoprint and klipper: 280x250 and origin of bed set at lower left.

In the first post I was using G90 in the START_PRINT macro, I removed it as it is not added by default by cura.

The current behaviour is that after homing it's prepping the extruder and then it gets stuck, no error, after I hit cancel print it reports: Move exceeds maximum extrusion.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4302321/klippy.log)

[testprint2.gcode.txt](https://github.com/KevinOConnor/klipper/files/4302304/testprint2.gcode.txt)

ld3300, 2020-03-07:
You definitely need to add G90 back in.  That would cause some problems.  I would put it in the Cura G-code before START_PRINT.  I'm not sure why Cura wouldn't add that automatically. Klipper is seeing all the XYZ moves as being relative, which would cause problems, but something else weird is happening first.

I see in the log where you have your START_PRINT macro, then it pushes the extruder 13mm, but probably too fast for your hot end. Then there is a line move in there that makes it look like the G-code tried to jump a few thousand lines into the print file which makes it try to move while extruding 400mm of filament.  Do you have any post processing scripts in Cura? Can you copy your startup gcode in Cura?  Can you also download the version of the g-code file from octoprint and send that as well (in case there is a plugin or something in octoprint that is changing the gcode).

insidecircuits, 2020-03-08:
I added back G90 right after homing. The start and end g-code are macros in klipper, so you can see them in the config. In cura I just call START_PRINT and END_PRINT.
The two gcode files are identical, no post processing happening.
I disabled most of my octoprint plugins that could interfere with gcode, but still nothing.

[image]

[klippy.log](https://github.com/KevinOConnor/klipper/files/4302957/klippy.log)

[octoprint-testprint3.gcode.txt](https://github.com/KevinOConnor/klipper/files/4302954/octoprint-testprint3.gcode.txt)

[cura-testprint3.gcode.txt](https://github.com/KevinOConnor/klipper/files/4302952/cura-testprint3.gcode.txt)

insidecircuits, 2020-03-08:
I reinstalled octoprint and klipper, started fresh and now it seems to be okay.

closed 2020-03-08 by insidecircuits

ld3300, 2020-03-08:
Glad to hear, that was a weird one.