Klipper3d/klipper #2600: Lost communication with MCU 'mcu' (after adding new camera)

yoc1983, 2020-03-07:
I run multiple camera's on my 3B+ pi 
a pi camera(CSI) and microscope camera (USB)
i unplugged the pi cam and install a Logitech C920 running 1920x1080 10fps
since then the printer randomly pausing the print giving the
"Lost communication with MCU 'mcu'".

if i unplug the logitech going back to the pi cam it's all working great for hours.
i tried to put a better quality USB cable (UGREEN) between pi and my skr 1.3 board 
and also checked my power supply as noted in the FAQ but no success
i know people run several C920 with the same resolution successfully
any suggestion how to progress here?

[klippy.log](https://github.com/KevinOConnor/klipper/files/4302595/klippy.log)

kakou-fr, 2020-03-07:
I had the same problem last month, doing a "sudo dmesg -c" after lost communication show a reboot of the USB hub (rpi internal) due to EMI ... so I added so ferrite on the usb wire of the camera and now it looks ok since 15 days.

yoc1983, 2020-03-07:
@kakou-fr  thanks for the info,
the wire already got a ferrite on the USB connector side
did you add another or a different kind?

kakou-fr, 2020-03-08:
yes they are a small ferrite on the usb. i put two bigger ferrite.

yoc1983, 2020-03-08:
@kakou-fr 
i added 2 more ferrite to the one already came with the cable but got the same random stop
this time i also run 'sudo dmesg –c' 
it seems like the camera is disconnecting
causing also a restart to klipper??
this what 'sudo dmesg –c' gave:

[16800.396990] WARN::dwc_otg_hcd_handle_hc_fsm:2458: Unexpected IRQ state on FSM                                                                                                              transaction:dev_addr=17 ep=2 fsm=6, hcint=0x00000002

[16800.398392] usb 1-1.2: USB disconnect, device number 16
[16800.400374] usb 1-1.3: USB disconnect, device number 19
[16800.597106] Indeed it is in host mode hprt0 = 00001501
[16800.806091] usb 1-1: new high-speed USB device number 20 using dwc_otg
[16800.806223] Indeed it is in host mode hprt0 = 00001101
[16801.046223] usb 1-1: New USB device found, idVendor=0424, idProduct=2514
[16801.046232] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[16801.046641] hub 1-1:1.0: USB hub found
[16801.046689] hub 1-1:1.0: 4 ports detected
[16801.366039] usb 1-1.1: new high-speed USB device number 21 using dwc_otg
[16801.496211] usb 1-1.1: New USB device found, idVendor=0424, idProduct=2514
[16801.496220] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber                                                                                                          =0
[16801.496618] hub 1-1.1:1.0: USB hub found
[16801.496668] hub 1-1.1:1.0: 3 ports detected
[16801.796038] usb 1-1.2: new high-speed USB device number 22 using dwc_otg
[16802.026051] usb 1-1.1.3: new full-speed USB device number 23 using dwc_otg
[16802.160698] usb 1-1.1.3: New USB device found, idVendor=2341, idProduct=abcd
[16802.160706] usb 1-1.1.3: New USB device strings: Mfr=1, Product=2, SerialNumb                                                                                                             er=3
[16802.160710] usb 1-1.1.3: Product: lpc1768
[16802.160714] usb 1-1.1.3: Manufacturer: Klipper
[16802.160718] usb 1-1.1.3: SerialNumber: 04000016E9A028AF1361C25AC32000F5
[16802.161336] cdc_acm 1-1.1.3:1.0: ttyACM0: USB ACM device
[16802.466077] usb 1-1.1.1: new high-speed USB device number 24 using dwc_otg
[16802.596310] usb 1-1.1.1: New USB device found, idVendor=0424, idProduct=7800
[16802.596319] usb 1-1.1.1: New USB device strings: Mfr=0, Product=0, SerialNumb                                                                                                             er=0
[16802.863973] libphy: lan78xx-mdiobus: probed
[16802.914005] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[16804.227821] usb 1-1.2: New USB device found, idVendor=046d, idProduct=082d
[16804.227829] usb 1-1.2: New USB device strings: Mfr=0, Product=2, SerialNumber                                                                                                             =1
[16804.227833] usb 1-1.2: Product: HD Pro Webcam C920
[16804.227838] usb 1-1.2: SerialNumber: 5DDF092F
[16804.231338] uvcvideo: Found UVC 1.00 device HD Pro Webcam C920 (046d:082d)
[16804.232203] uvcvideo 1-1.2:1.0: Entity type for entity Processing 3 was not i                                                                                                        nitialized!
[16804.232211] uvcvideo 1-1.2:1.0: Entity type for entity Extension 6 was not in                                                                                                             itialized!
[16804.232216] uvcvideo 1-1.2:1.0: Entity type for entity Extension 12 was not i                                                                                                             nitialized!
[16804.232221] uvcvideo 1-1.2:1.0: Entity type for entity Camera 1 was not initi                                                                                                             alized!
[16804.232225] uvcvideo 1-1.2:1.0: Entity type for entity Extension 8 was not in                                                                                                             itialized!
[16804.232230] uvcvideo 1-1.2:1.0: Entity type for entity Extension 9 was not in                                                                                                             itialized!
[16804.232234] uvcvideo 1-1.2:1.0: Entity type for entity Extension 10 was not i                                                                                                             nitialized!
[16804.232239] uvcvideo 1-1.2:1.0: Entity type for entity Extension 11 was not i                                                                                                             nitialized!
[16804.232423] input: HD Pro Webcam C920 as /devices/platform/soc/3f980000.usb/u                                                                                                             sb1/1-1/1-1.2/1-1.2:1.0/input/input6
[16804.706059] usb 1-1.3: new high-speed USB device number 25 using dwc_otg
[16804.916526] usb 1-1.3: New USB device found, idVendor=1908, idProduct=2311
[16804.916534] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber                                                                                                             =0
[16804.916538] usb 1-1.3: Product: USB2.0 PC CAMERA
[16804.916542] usb 1-1.3: Manufacturer: Generic
[16804.917098] uvcvideo: Found UVC 1.00 device USB2.0 PC CAMERA (1908:2311)
[... cut ...]
[16804.917490] uvcvideo 1-1.3:1.0: Entity type for entity Camera 1 was not initi                                                                                                             alized!
[16804.917657] input: USB2.0 PC CAMERA: USB2.0 PC CAM as /devices/platform/soc/3                                                                                                             f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/input/input7
[16805.566393] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[16830.646651] usb 1-1.2: reset high-speed USB device number 22 using dwc_otg

kakou-fr, 2020-03-08:
so it's not the same issue than me, i have this error un dmesg : 
USB port nn disabled by hub (EMI?), re-enabling

maybe a lake of power ? did you use a power rated 2.5A ?

yoc1983, 2020-03-08:
@kakou-fr 
I am using my phone adapter which is rated max 4A

superjeng1, 2020-03-11:
Hi, I had a similar issue. When I use klipper and a webcam at the same time, the Raspberry PI will struggle and eventually fail. I suppose it is a io issue, since Raspberry PI runs on SD cards. After some digging around, I found that it's klipper's log using too much resources. So I attempted to write it to ramdisk, and here we go, it works! Basically, just mount `/tmp` as a ramdisk in `/etc/fstab` and reboot, things should work fine then.

KevinOConnor, 2020-03-21:
Okay, thanks for the info.  However, if the usb camera is interfering with the usb printer connection, then there isn't much we can be done about it.  It is, of course, some type of hardware issue.

-Kevin

yoc1983, 2020-03-25:
@KevinOConnor thanks..
I still got errors without cam now but less errors
so i reinstall everything with octopi 0.17
then i noticed that octopi 0.16 and above are recommended for Klipper
i used 0.15 before
for now i got a few prints without any error (no camera)
will have to continue testing but for now it looks solid

KevinOConnor, 2020-04-08:
I'm going to close this as it looks like a solution was found for the original issue.

-Kevin

closed 2020-04-08 by KevinOConnor