Klipper3d/klipper #2604: Internal error during connect: MCU_endstop instance has no attribute 'TimeoutError'

SparkGuilty, 2020-03-08:
[serial.log](https://github.com/KevinOConnor/klipper/files/4302929/serial.log)
[octoprint.log](https://github.com/KevinOConnor/klipper/files/4302930/octoprint.log)
[printer.cfg.pdf](https://github.com/KevinOConnor/klipper/files/4302941/printer.cfg.pdf)
[klippy.zip](https://github.com/KevinOConnor/klipper/files/4302947/klippy.zip)

HELP!

While trouble shooting issues with my BL Touch, I was successful in getting it to deploy and retract but could not get the endstop function to work. It would always flash red and continue to move towards the bed after executing a G28 function and manually triggering the probe with my finger. After some trial and error within the config file and z+, z- connection on my mainboard (SKR V1.3) I began receiving the error " Internal error during connect: MCU_endstop instance has no attribute 'TimeoutError'". Nothing I do will get rid of it and I've run out of things to try. Any help with this matter will be greatly appreciated.

ld3300, 2020-03-08:
Have you tried it on a non-modified install of Klipper?  The install you have has had code level changes to it so is possibly missing a file or has had function changes that are not compatible with that version.
I recommend force updating to the master branch and trying again.

SparkGuilty, 2020-03-08:
Thank you! 

Reinstalling Kipper seemed to fix the issue, so I'm back to a half working BL Touch. Can anyone assist in getting the firmware to recognize when the probe is triggered during a z homing movement? 
I have been reading up on adding a pull-up resistor but unsure what size or exactly how the install should go.

ld3300, 2020-03-08:
@SparkGuilty I'm glad that worked.  I'm sorry the BL touch is struggling.  There are a couple of other issues on here that have been active in the last couple of weeks where it's been discussed.  I'm not sure if any of them have contributions that can help, but do a search in the repo and sort the issues by date.  Hopefully one of the recent conversations can help you get it going.

KevinOConnor, 2020-03-21:
I'm going to close this as it looks like a solution was found for the original issue.  With regard to the BL-Touch, the best information we have is in docs/BL-Touch.md.

-Kevin

closed 2020-03-21 by KevinOConnor