Klipper3d/klipper #2608: SKR 1.1 + TMC5160 drivers + March Commit = motor clicks and skips

johngianni, 2020-03-09:
Since the 5th of march release I'm experiencing some problems with my printer. Initially, the printer couldn't even complete the homing procedure, but after the weekend I've updated Klipper to the latest commit and now the printer works, but the X and Y motors are emitting click and slight skips noise while moving. 
Reverting back to the 4th of march commit fixes all the issues, so I suppose that the latest features regarding stm32 and stuff should be checked.
I don't think that the logs would provide any information aside of the config file, as no error is shown during the clicks and skips. 
Please help me to send you the correct info required to debug. 
 
[klippy.log](https://github.com/KevinOConnor/klipper/files/4308478/klippy.log)

KevinOConnor, 2020-03-11:
It would help if you could update to the latest code and run DUMP_TMC on a stepper that is producing noise.  Then provide the log file from that session.  Then change to the previous working version, run DUMP_TMC on the same stepper, and then provide that log file.  Attach both logs here.

-Kevin

johngianni, 2020-03-11:
Thx for getting back. Here are the two logs, and below the differences in the TMC dump:

4th of march commit: 
`========== Write-only registers ==========
TPWMTHRS:   0000001f TPWMTHRS=31
COOLCONF:   00050000 sgt=5
IHOLD_IRUN: 00060704 IHOLD=4 IRUN=7 IHOLDDELAY=6
PWMCONF:    c40d001e PWM_OFS=30 pwm_freq=1 pwm_autoscale=1 pwm_autograd=1 PWM_REG=4 PWM_LIM=12
TPOWERDOWN: 0000000a TPOWERDOWN=10
========== Queried registers ==========
GCONF:      00000004 en_pwm_mode=1
CHOPCONF:   14410153 toff=3 hstrt=5 hend=2 tbl=2 tpfd=4 MRES=4(16usteps) intpol=1
GSTAT:      00000005 reset=1(reset) uv_cp=1(Undervoltage!)
DRV_STATUS: 00074000 stealth=1 CSACTUAL=7
FACTORY_CONF: 0000000c FACTORY_CONF=12
IOIN:       30000040 SD_MODE=1 VERSION=0x30
LOST_STEPS: 00000000
MSCNT:      00000235 MSCNT=565
MSCURACT:   012d017d CUR_A=-131 CUR_B=-211
OTP_READ:   0000000c OTP_FCLKTRIM=12
PWM_SCALE:  01f20040 PWM_SCALE_SUM=64 PWM_SCALE_AUTO=-14
PWM_AUTO:   002d00c7 PWM_OFS_AUTO=199 PWM_GRAD_AUTO=45
TSTEP:      0000018b TSTEP=395`

Lastest MASTER Commit
`========== Write-only registers ==========
TPWMTHRS:   0000001f TPWMTHRS=31
COOLCONF:   00050000 sgt=5
GLOBALSCALER: 00000043 GLOBALSCALER=67
IHOLD_IRUN: 00061f13 IHOLD=19 IRUN=31 IHOLDDELAY=6
PWMCONF:    c40d001e PWM_OFS=30 pwm_freq=1 pwm_autoscale=1 pwm_autograd=1 PWM_REG=4 PWM_LIM=12
TPOWERDOWN: 0000000a TPOWERDOWN=10
========== Queried registers ==========
GCONF:      00000004 en_pwm_mode=1
CHOPCONF:   14410153 toff=3 hstrt=5 hend=2 tbl=2 tpfd=4 MRES=4(16usteps) intpol=1
GSTAT:      00000005 reset=1(reset) uv_cp=1(Undervoltage!)
DRV_STATUS: 001f4000 stealth=1 CSACTUAL=31
FACTORY_CONF: 0000000c FACTORY_CONF=12
IOIN:       30000040 SD_MODE=1 VERSION=0x30
LOST_STEPS: 00000000
MSCNT:      00000378 MSCNT=888
MSCURACT:   00cf0179 CUR_A=-135 CUR_B=207
OTP_READ:   0000000c OTP_FCLKTRIM=12
PWM_SCALE:  01f90041 PWM_SCALE_SUM=65 PWM_SCALE_AUTO=-7
PWM_AUTO:   001e0033 PWM_OFS_AUTO=51 PWM_GRAD_AUTO=30
TSTEP:      00000168 TSTEP=360`

Can't help reading those values,

though.... 
[klippy_master.zip](https://github.com/KevinOConnor/klipper/files/4317360/klippy_master.zip)

KevinOConnor, 2020-03-11:
The change in behaviour is likely due to PR #2534 (aka commit d6c2e24f).  However, the new settings should, in general, be an improvement - your old configuration led to a setting of IRUN=7 which could lead to "coarser micro-steps" on the driver.

If you're getting poor results with the new code, I recommend trying to tune the run_current settings.  A higher (or perhaps lower) run_current may reduce the noise on the axis.  You may also want to double check that the sense_resistor setting is correct.

-Kevin

johngianni, 2020-03-11:
Having tuned those parameters fixed it... thank you very much for the wonderful support...

closed 2020-03-11 by johngianni

KevinOConnor, 2020-03-15:
Thanks.  For future reference, what were the new settings that improved the results for you?

-Kevin

johngianni, 2020-03-19:
Hi sure,
increased the Vref by 0.1 and also used 1/32 microsteps