Klipper3d/klipper #2609: Does anybody have issues with the latest Octoprint?

sedlacekdavid, 2020-03-09:
Hi, it seems that my RPi3b doesn't keep up with octoprint anymore. I've upgraded to the latest Klipper and Octoprint, they announced it a few days ago.

Does anybody here have similar issue? 

In my case it consumes 100 % of cpu and hangs from time to time for seconds. Even on direct lines like bridges.

If I use Klipper in virtual SD card mode, it works 100 % fine and with much less cpu consumption:
[image]

sedlacekdavid, 2020-03-09:
This is the result, blobs are the places where it hangs for seconds

[image]

Afterschnitzel, 2020-03-09:
Same here with Raspberry Pi2 and SKR Pro v1.1
I had to uninstall almost all plugins to print constantly without unwanted pauses.

johngianni, 2020-03-10:
Are you using Python 3.6 o 2? With 3.6 I have no issues, Pi2 and SKR Pro...

sedlacekdavid, 2020-03-10:
It's Python3+ .... I also haven't got any issues until I upgraded. I guess we should reach out to them : ) It seems there's nothing really with Klipper itself I think.

Afterschnitzel, 2020-03-10:
> Are you using Python 3.6 o 2? With 3.6 I have no issues, Pi2 and SKR Pro...

I will try this afternoon. I have no issue with only two plugin - Bed Visualiser and Themeify. To have more plugins are problematic.

sedlacekdavid, 2020-03-10:
I'd understand having issues with an ancient RPi1/2, but not with 3b+.

closed 2020-03-10 by sedlacekdavid