Klipper3d/klipper #2611: M73 command results in strange display output

godefroi, 2020-03-12:
I'm using an Octoprint plugin to pass M73 GCode commands to Klipper during prints. This, as expected, causes the progress bar to show, but the times the show up to the right of the bar are... nonsensical.

There are actually two plugins in play here, but I don't think they would interact in any way on the Klipper side. One sends M117 commands to show the current layer and Z height. This works great. The other sends M73 commands to show completion percent. This doesn't. The end result is this:

[image]

This is near the end of a 7-hour 45-minute print. I don't know what "00:13" is meant to convey, but probably not that. The other number that flashes there is "-00:00" which is probably roughly correct; there was no more than a minute or two left when I took the picture. Sometimes, however, it's wildly incorrect.

```

ld3300, 2020-03-13:
That is supposed to be the current printing time. , you are correct that the negative should be the estimated remaining time.  If I remember right mine was doing weird things too.  It seems to be working now, I wonder if it was fixed sometime in the last 2 months.  There were some display changes in March that might have been what made mine work correctly.

KevinOConnor, 2020-03-15:
Unfortunately, the display's print time estimate is not particularly accurate.  There is no g-code command to indicate the "start of a print" nor "end of a print", so Klipper can only estimate how long a print has been in progress.  It looks like this particular print is suffering from "print stalls" - the printer is moving faster than OctoPrint can send commands.  That is causing the print time estimate to reset.  The very latest display code (as of a few days ago) may improve this.  Using virtual_sdcard may also improve this (as well as reduce pauses during the print due to OctoPrint running slow).  See config/example-extras.cfg for the details.

-Kevin

godefroi, 2020-03-16:
The M73 Progress plugin for Octoprint (which I'm using) has the ability to send the estimated remaining time as another parameter (`R` I think); if Klipper could support that, then it would fix at least one of the two issues here (one being elapsed time being reset, the other being remaining time being calculated incorrectly).

https://plugins.octoprint.org/plugins/m73progress/

ld3300, 2020-03-17:
It looks like it wouldn't be too hard to implement.  I will give it a try when I get some time and hopefully Kevin will have some time to look it over and check it.  Might be a couple of days.  The question I have is should "P" always be required with "R" as optional, or both as optional parameters.  It looks like the plugin sends P and R.

godefroi, 2020-03-17:
The plugin always sends P, can be configured to also send R. Looking around, I don't see any firmware that supports R but not P.

ld3300, 2020-03-18:
I think it would be less error prone that way

ld3300, 2020-03-26:
see PR #2646

KevinOConnor, 2020-04-24:
I'm going to close this issue as it looks like the original question was answered (Klipper's estimates are limited).  There's some improvement that can be made to the display - lets track that in #2646 .

-Kevin

closed 2020-04-24 by KevinOConnor