Klipper3d/klipper #2612: Alfawise U30 (SMT32F103) original firmware backup before flash

elegos, 2020-03-12:
Hello!

I'd like to try the firmware, but I'd really like to backup the original one before. As the script is able to flash the firmware on the machine, I suppose you know a way to extract the firmware from it. I'd like to be able to go back to the original one, in case of need.

May you please help me? Thanks :)

KevinOConnor, 2020-03-15:
Unfortunately, every chip bootloader is different.  I don't know what steps would be needed to backup the chip on your particular board.  There is some general information at: https://www.klipper3d.org/Bootloaders.html

-Kevin

elegos, 2020-03-16:
I figured out that this printer won't overwrite the bootloader but only the firmware, and as it can flash the firmware from the sdcard at boot time, there is no need to backup (original firmware is downloadable) :)

closed 2020-03-16 by elegos

ulysse132, 2020-10-02:
@elegos could you explain me how you flashed klipper on the alfawise please ? In the doc, I only see flash through usb. Thanks a lot for your answer !

elegos, 2020-10-02:
Hello @ulysse132 

In the end I actually flashed the ROM via microSD, after correctly compiling it.

ulysse132, 2020-10-03:
Hello @elegos 
How did you flashed it ? You just put klipper.bin on the root of the sd card ? Did you have to rename it to project.bin ? What about longer3d.ui which is needed to update the alfawise ?

elegos, 2020-10-05:
Just rename it in project.bin and you should be fine :)

ulysse132, 2020-10-12:
Thanks a lot for your help ! It works like a charm ! I just reduced by half the global acceleration and I can print at 120 mm/s without any issue