Klipper3d/klipper #2613: Alfawise U30 (2018) connection timeout

elegos, 2020-03-12:
Hello!
I've decided to try out Klipper even though I was not able to backup my original firmware.

The Alfawise U30 printer uses a STM32F103VET6 micocontroller. It flashes the firmware at boot time if there is a `project.bin` file in the root of the sdcard. Indeed when I tried to flash via USB (`make flash FLASHDEVICE=/dev/ttyUSB0`) the following error appeared:

```
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
dfu-util: No DFU capable USB device available

Failed to flash to /dev/ttyUSB0: Error running dfu-util

If the device is already in bootloader mode it can be flashed with the
following command:
  make flash FLASH_DEVICE=1eaf:0003

If attempting to flash via 3.3V serial, then use:
  make serialflash FLASH_DEVICE=/dev/ttyUSB0

src/stm32/Makefile:56: recipe for target 'flash' failed
make: *** [flash] Error 255
```

menuconfig options:
```
Micro-controller: STMicroelectronics STM32
Processor model: STM32F103
Bootloader offset: no bootloader
Use USB for communication (instead of serial): checked
USB ids: (left unchanged)
```

I've thus copied `out/klipper.bin` in the sdcard as `project.bin`. At the printer's startup a progress line showed up. After around one minute (progress line 100% by a while), I restarted the printer. Nothing showed up on the monitor (I suspect that because of the minimal firmware?) but a white rectangle at the edges of the monitor itself.

I've then went to the raspberry installation and copied the Creality Ender 3 profile to start from because this seems pretty much a clone of it.

I've then restarted klipper, but the following error raised:
```
Starting serial connect
Timeout on serial connect
Wait for identify_response
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/serialhdl.py", line 67, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/pi/klipper/klippy/serialhdl.py", line 182, in send_with_response
    return src.get_response(cmd, self.default_cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 238, in get_response
    self.serial.raw_send_wait_ack(cmd, minclock, minclock, cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 174, in raw_send_wait_ack
    raise error("Serial connection closed")
error: Serial connection closed
```

Baud rate 115200 is the same I used with the original firmware, I don't know if it's the correct one, or which one I should choose... or even if the installation went fine at all!

May you please help me? Thanks

KevinOConnor, 2020-03-15:
If you programmed the chip via an sd card, then it has a bootloader and it will be necessary to specify a bootloader offset.  (Unfortunately, the offset is rarely listed in a convenient location - if the printer runs Marlin, maybe you can find it from the Marlin source.)  If the printer is using native USB then it should change ids when you install Klipper successfully (you will need to rerun `ls /dev/serial/by-id/*`).  If the printer is using an external serial converter chip then you'll need to identify which stm32f103 serial port is in use - always leave the baud rate at the default of 250000.

-Kevin

elegos, 2020-03-16:
I've managed to make the firmware run thanks to @berghauz's help. I've then took his branch and updated to master: https://github.com/elegos/klipper/tree/mks-robin-support

I'll push the Alfawise U30 configuration too (which is the U20 one, too, but the printable area). I see that @berghauz's branch is older than the STM32 support. I know little to nothing about this kind of programming, so you (or someone else, or myself with your help?) might standardise the code and merge it in the original's master.

KevinOConnor, 2020-03-21:
Perhaps we should merge this ticket into #2397 then.

It would be great to get the necessary support (and example configs) into the main Klipper repo.

-Kevin

thucar, 2020-03-21:
That would be great. I've now tried getting the Klipper master branch as well as @elegos's and @berghauz's branches to work with self compiled firmware (running the bin through mks_robin_xor.py scritp) but no luck. The only working solution for me right now is using @berghauz's precompiled binary.

berghauz, 2020-03-21:
@thucar if you are trying to make binary for alfawise you should'nt use mks_robin_xor.py, because alfawise bootloader did not use encryption (and it is not msk board, even it looks veeery similar) and for shure you need to pull updated mks-robin-support branch, alfawise (vtor offset support) was added a week ago.

thucar, 2020-03-21:
@berghauz I'm using the MKS Robin Nano board and I did try the latest mks-robin-support branch. I think I tried all the different option combinations in the menuconfig but none of them worked for me. Also not sure if related, STM32F103xB Processor model fails to compile for me. Make ends in an error:

```
Build Kconfig config file
  Creating symbolic link out/board
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/initial_pins.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/thermocouple.o
  Compiling out/src/i2ccmds.o
  Compiling out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
  Compiling out/src/buttons.o
  Compiling out/src/tmcuart.o
  Compiling out/src/spi_software.o
  Compiling out/src/neopixel.o
  Compiling out/src/stm32/watchdog.o
  Compiling out/src/stm32/gpio.o
  Compiling out/src/generic/crc16_ccitt.o
  Compiling out/src/generic/armcm_boot.o
  Compiling out/src/generic/armcm_irq.o
  Compiling out/src/generic/armcm_reset.o
  Compiling out/src/../lib/stm32f1/system_stm32f1xx.o
  Compiling out/src/stm32/stm32f1.o
src/stm32/stm32f1.c: In function 'armcm_main':
src/stm32/stm32f1.c:189:18: error: 'RCC_AHBENR_FSMCEN' undeclared (first use in this function)
  RCC->AHBENR &= ~RCC_AHBENR_FSMCEN;
                  ^
src/stm32/stm32f1.c:189:18: note: each undeclared identifier is reported only once for each function it appears in
src/stm32/stm32f1.c:192:18: error: 'RCC_AHBENR_DMA2EN' undeclared (first use in this function)
  RCC->AHBENR &= ~RCC_AHBENR_DMA2EN;
                  ^
src/stm32/stm32f1.c:195:18: error: 'RCC_AHBENR_SDIOEN' undeclared (first use in this function)
  RCC->AHBENR &= ~RCC_AHBENR_SDIOEN;
                  ^
Makefile:66: recipe for target 'out/src/stm32/stm32f1.o' failed
make: *** [out/src/stm32/stm32f1.o] Error 1
```

berghauz, 2020-03-21:
@thucar 
1. STM32F103xB != STM32F103xE, xB does not have fsmc bus and much more, so there is no surprise it won't compiling.
2. "but none of them worked for me" tells absolutely nothing for me, give me please some details, at least what exactly not working and how it's looks.

but before try my .config https://pastebin.com/gY1QxU0E and 
make && make robin_nano.bin
i bet the problem is near it.
btw: robin mini and nano hardware absolutely equal for mcu part of klipper, but different from configuration point.
btw2: if you have 3.5'' display (sapphire pro printer, for instance) you need to append a suffix to file name, like robin_nano35.bin

thucar, 2020-03-21:
Tried your conf. It compiles fine, flashes to the printer also fine but Klipper wont be able to connect to the printer.

```
=======================
Extruder max_extrude_ratio=0.415752
Starting serial connect
Timeout on serial connect
Wait for identify_response
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/serialhdl.py", line 67, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/pi/klipper/klippy/serialhdl.py", line 182, in send_with_response
    return src.get_response(cmd, self.default_cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 238, in get_response
    self.serial.raw_send_wait_ack(cmd, minclock, minclock, cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 174, in raw_send_wait_ack
    raise error("Serial connection closed")
error: Serial connection closed
```

KevinOConnor, 2020-03-21:
@berghauz - for the Alfawise, is it just the 64KiB bootloader offset and resetting some peripheral clocks?

@elegos - can you try with the code on the work-stm32f1-20200321 branch?  (Be sure to select the 64KiB bootloader during make menuconfig.)

`cd ~/klipper ; git fetch ; git checkout origin/work-stm32f1-20200321 ; sudo service klipper stop ; make menuconfig ; make ; make flash ; sudo service klipper start`

-Kevin

berghauz, 2020-03-21:
@KevinOConnor yep, encryption also not needed. fsmc clock should be in reset state in case you need to use iic bus (fsmc and iic shares common pins) if not, then fsmc may be leave as is after bootloader, but screen will show "ready" and backlight will light on :)
@thucar i have mks robin nano somewhere in a stash, will check it soon

KevinOConnor, 2020-03-21:
@berghauz 

> fsmc clock should be in reset state in case you need to use iic bus

It should be safe to reset AHBENR on all boards, as in commit cafa7420 (on work-stm32f1-20200321).

> backlight will light on

It should be possible to turn off the backlight with an output_pin declaration in the printer.cfg file (or alternatively via the "GPIO pins to set at micro-controller startup" under the low-level options of make menuconfig).

Thanks,
-Kevin

elegos, 2020-03-21:
@thucar using my mks-robin-support's branch, in order to make a U20/U30 machine run the firmware, you have to compile with the following options:

- Enable extra low-level configuration options: YES
- Micro-controller Architecture: STMicroelectronics STM32
- Processor model: STM32F103xE
- Bootloader offset: 64KiB bootloader (Alfawise U20/30)
- Clock Reference: 8 MHz crystal
- Use USB for communication (instead of serial): NO
- Serial Port: USART1
- Specify a custom step pulse duration: NO
- GPIO pins to set at micro-controller startup (none)
- Is STM32F1 are MKS Robin family board? NO
- Is STM32F1 are Alfawise U20/30 board? YES

@KevinOConnor I've tried the new branch with the following options:

- Enable extra low-level configuration options: YES
- Micro-controller Architecture: STMicroelectronics STM32
- Processor model: STM32F103
- Bootloader offset: 64KiB bootloader (Alfawise)
- Clock Reference: 8 MHz crystal
- Use USB for communication (instead of serial): NO
- Serial Port: USART1
- Baud rate for serial port: 250000
- Specify a custom step pulse duration: NO
- GPIO pins to set at micro-controller startup (none)

The result is an unbootable firmware.

P.S.
I've uploaded on my branch the U20 configuration file :)

KevinOConnor, 2020-03-21:
Oops - there was a dumb error in the work-stm32f1-20200321 branch.  @elegos - could you try again?

`cd ~/klipper ; git fetch ; git checkout origin/work-stm32f1-20200321 ; sudo service klipper stop ; make menuconfig ; make ; make flash ; sudo service klipper start`

-Kevin

elegos, 2020-03-22:
@KevinOConnor nope, unfortunatelly the firmware won't still boot.

P.S.
`make flash` can't be used here, you need to put it on the sdcard :) (unless you've got a STLink hardware, I suppose)

KevinOConnor, 2020-03-23:
@elegos - that's odd.  Can you grab the latest code, do a full rebuild, and retry.

`cd ~/klipper ; git fetch ; git checkout origin/work-stm32f1-20200321 ; sudo service klipper stop ; make menuconfig ; make ; make flash ; sudo service klipper start`

If it fails, please let the code do its retries for a few minutes and then attach the resulting Klipper log here.  If it fails, can you then select the new "Force ram size of 0xffd8" in make menuconfig, do a full rebuild, and retry?  Be sure to wait a few minutes and attach that log file as well.

(Yes, I know "make flash" doesn't directly work - I include it as a reminder for others that may not have read through this full thread.)

-Kevin

elegos, 2020-03-24:
Ok, in both the cases the following happens (with or without the forced RAM size):

1. flash the firmware. The screen won't reset (i.e. the progress bar will remain).
2. restart klipper: it's able to get data from the printer (i.e. the firmware booted nonetheless).
3. shutdown the printer.
4. startup the printer. The printer's screen is ON with no data on it.
5. klipper hangs (no temp data will arrive, no command can be sent).

[klippy.log](https://github.com/KevinOConnor/klipper/files/4374275/klippy.log)

KevinOConnor, 2020-03-24:
Add `restart_method: command` to the `[mcu]` config section (see config/example.cfg for the details).  Also, note that if you manually reset the printer (not via the Klipper host software) then you will need to issue a FIRMWARE_RESTART command to clear the error state.

If you want to reset the display, it looks like you would add the following to the printer.cfg file:
```
[static_digital_output display_reset]
pins: !PC6, !PD13
```
(Or, set the equivalent in the "GPIO pins to set at micro-controller startup" section of "make menuconfig" - it's in the low-level options.)

-Kevin

KevinOConnor, 2020-03-26:
Is the Alfawise board now working on Klipper?  If so, it would be great to get an example config file added to the repo.

-Kevin

elegos, 2020-03-30:
Hello @KevinOConnor, sorry for the delay, but had hard time here.

The solution seems to work now. Follows the .config and printer.cfg.

[Alfawise U20-30 configuration.zip](https://github.com/KevinOConnor/klipper/files/4402034/Alfawise.U20-30.configuration.zip)

P.S.
Is it normal that when you turn off the printer and back again you have to manually do a firmware_restart?

KevinOConnor, 2020-03-31:
Okay, thanks.  I added a starting config (commit d6034013).

Yes, it is normal to require a FIRMWARE_RESTART if the micro-controller is manually restarted.  The host software is not expecting the micro-controller to "disappear", so that command is needed to clear the resulting error state.

To your earlier question about the display - alas, I don't know about the touch screen.

Stay safe,
-Kevin

elegos, 2020-03-31:
It would come handy to support in the menuconfig a first option to pre-select the right settings for known printers, so that a user don't have to know which chipset and particular settings are required for his/her printer... i.e. pre-load from known .config files.

KevinOConnor, 2020-04-08:
I don't know enough about the display to comment on what would be required to support it.  For those displays that are supported, the content is configurable - see config/example-extras.cfg for the details.

-Kevin

KevinOConnor, 2020-04-15:
Okay, lets close this as it looks like the original issue has been solved.

-Kevin

closed 2020-04-15 by KevinOConnor

Sinclair5, 2021-05-30:
hi,  i recently go klipper to work ,  but if the part cooling fan isn't set to the max  in my slicer,  it wont work at all during the print