Klipper3d/klipper #2614: USB connection between Pi and Maestro borked.

Grimpster, 2020-03-12:
Finally got back to install klipper on maestro, but then this happened.

I've tried several versions Buster and similar, the normal installation of klipper and DWC, and the manu7irls combined script, all leading back to the same result, I'm out of options here.

So when I'm at the point of flashing the MCU I'm doing:

running:
 ls /dev/serial/by-id/*
gives me:
/dev/serial/by-id/usb-Duet3D_Duet-if00

adding that to klipper.cfg along with the other things:
[mcu]
serial: /dev/serial/by-id/usb-Duet3D_Duet-if00

[virtual_sdcard]
path: /home/pi/sdcard

[web_dwc2]
## optional - defaulting to Klipper
printer_name: Fizban
# optional - defaulting to 0.0.0.0
listen_adress: 0.0.0.0
# needed - use above 1024 as nonroot
listen_port: 4750
# optional defaulting to dwc2/web. Its a folder relative to your virtual sdcard.
web_path: dwc2/web

and when running the flash it looks like this:
blahblahblah
  Compiling out/src/generic/usb_cdc.o
  Compiling out/src/atsam/adc.o
  Compiling out/src/atsam/sam4s_sysinit.o
  Building out/compile_time_request.o
Version: v0.8.0-366-g986e9c88-dirty-20200312_135751-raspberrypi
  Preprocessing out/src/generic/armcm_link.ld
  Linking out/klipper.elf
  Creating bin file out/klipper.bin
  Flashing out/klipper.bin to /dev/serial/by-id/usb-Duet3D_Duet-if00
Entering bootloader on /dev/serial/by-path/platform-3f980000.usb-usb-0:1.3:1.0
Device reconnect on /dev/serial/by-path/platform-3f980000.usb-usb-0:1.3:1.0
lib/bossac/bin/bossac -U -p /dev/serial/by-path/platform-3f980000.usb-usb-0:1.3:1.0 -e -w out/klipper.bin -v

No device found on /dev/serial/by-path/platform-3f980000.usb-usb-0:1.3:1.0
Failed to flash to /dev/serial/by-id/usb-Duet3D_Duet-if00: Error running bossac
make: *** [src/atsam/Makefile:54: flash] Error 255

What gives? I tried giving read/write rights to the user, a german site had a similar issue refereing to case sensitive files, but I can't seem to find anything like that.
A guy on the discord mentioned a recent change in klippers USB handling? - could this be the culprint ?

[Klippy.log](https://github.com/KevinOConnor/klipper/files/4326126/Klippy.log)

ld3300, 2020-03-13:
Hi @Grimpster,

Try a clean install of Klipper without DWC and see if it works.

Grimpster, 2020-03-13:
Fixed my issue, thanks.
A copy/paste from the klipper discord. Hoping it get's pinned, afaik Manu7irl is adding the info to his script at least.

---Installing Klipper/DWC2 on Duet Maestro board---

This really took some effort for me, being used to do it on ramps, and finding it super easy! knowing the Maestro took some effort from variuos dev's to fix, and I'm sure it gave them some long hairy nights. It certainly did for me, trying to make it run klipper and DWC.

For anyone having issues, or wanting to install it on the Maestro here's what I think would help you out loads!

Needed:
Maestro board
Raspi
a sdcard with Buster (or whatever linux you like to run I guess)
A jumper! <- this was my culprint!

#1 setup the Pi like you normally would following any of the guides out there.

#2 erase that maestro board! there's a jumper port you need to jump. Turn of the Maestro, plug it, turn it on wait 5 sec, remove jumper.

#3 I installed Klipper and DWC using @Manu7irl script https://github.com/manu7irl/klipper-DWC2-installer
But you can use the reps the script is pulling, from the official site https://klipper.info/klipper-+-dwc2-1

#4 This is where it got tricky for me, when being asked to do:
ls /dev/serial/by-id/*
You get an ID, The first ID! is for the flash only!!! it could look something like this:
/dev/serial/by-id/usb-03eb_6124-if00
then do:
make flash FLASH_DEVICE=/dev/serial/by-id/usb-03eb_6124-if00
When flashed, reboot the Maestro (Or the entire system as mines hooked up on the same reboot relay)
do another:
ls /dev/serial/by-id/*
output could look like:
/dev/serial/by-id/usb-Klipper_sam4s8c__dev_serial_by-id_usb-Duet3D_Duet-if00-if00
Copy this and enter it into the printer.cfg:
[mcu]
serial: /dev/serial/by-id/usb-Klipper_sam4s8c__dev_serial_by-id_usb-Duet3D_Duet-if00-if00

And now you should have Klipper and DWC2 working.
All you got to do now is setup the printer.cfg to match whatever printer you're running and you personal preferences.

closed 2020-03-13 by Grimpster