Klipper3d/klipper #2616: Trigorilla (2560)  with TMC2130 in SPI - No communication

robustini, 2020-03-14:
I installed the TMC2130 drivers in a friend's Anycubic Linear Plus (Delta), with "Trigorilla V1.1" board.
I had already done it in his time with Marlin and everything worked perfectly, today I'm trying with Klipper but the drivers don't communicate in SPI.
The wiring is respected as from the pdf that I attach, the drivers are correctly configured at the hardware level for the SPI, I don't understand why they don't work.

[image]
[image]
[image]
[klippy.log](https://github.com/KevinOConnor/klipper/files/4333011/klippy.log)
[tmc2130.on.Anycubic.kossel.trigorilla.pdf](https://github.com/KevinOConnor/klipper/files/4333012/tmc2130.on.Anycubic.kossel.trigorilla.pdf)

robustini, 2020-03-14:
16:01:59.869: // ========== Queried registers ==========
16:01:59.876: // GCONF: ffffffff I_scale_analog=1(ExtVREF) internal_Rsense=1 en_pwm_mode=1 enc_commutation=1 shaft=1(Reverse) diag0_error=1 diag0_otpw=1 diag0_stall=1 diag1_stall=1 diag1_index=1 diag1_onstate=1 diag1_steps_skipped=1 diag0_int_pushpull=1 diag1_pushpull=1 small_hysteresis=1 stop_enable=1 direct_mode=1 test_mode=1
16:01:59.881: // GSTAT: ffffffff reset=1 drv_err=1(ErrorShutdown!) uv_cp=1(Undervoltage!)
16:01:59.886: // IOIN: ffffffff STEP=1 DIR=1 DCEN_CFG4=1 DCIN_CFG5=1 DRV_ENN_CFG6=1 DCO=1 VERSION=0xff
16:01:59.891: // TSTEP: ffffffff TSTEP=1048575
16:01:59.897: // XDIRECT: ffffffff
16:01:59.902: // MSCNT: ffffffff MSCNT=1023
16:01:59.907: // MSCURACT: ffffffff CUR_A=-1 CUR_B=-1
16:01:59.913: // CHOPCONF: ffffffff toff=15 hstrt=7 hend=15 fd3=1 disfdcc=1 rndtf=1 chm=1 TBL=3 vsense=1 vhighfs=1 vhighchm=1 sync=15 MRES=15(0usteps) intpol=1 dedge=1 diss2g=1
16:01:59.920: // DRV_STATUS: ffffffff SG_RESULT=1023 fsactive=1 CS_ACTUAL=31 stallGuard=1 ot=1(OvertempError!) otpw=1(OvertempWarning!) s2ga=1(ShortToGND_A!) s2gb=1(ShortToGND_B!) ola=1(OpenLoad_A!) olb=1(OpenLoad_B!) stst=1
16:01:59.924: // PWM_SCALE: ffffffff PWM_SCALE=255
16:01:59.930: // LOST_STEPS: ffffffff LOST_STEPS=1048575
16:01:59.930: ok

robustini, 2020-03-14:
Solved.

closed 2020-03-14 by robustini

OttoKH, 2020-11-08:
What was the solution you ended up finding to this? I'm running into the exact same issue on my Mega-S trying to setup these new drivers.