Klipper3d/klipper #2617: Move a stepper (extruder) during an other manual_stepper

kakou-fr, 2020-03-14:
Hi,

To ease the load/unload of the filament in the extruder with the MMU2S, I want to move the extruder (constant speed) when the MMU2S (the manual_gear stepper) move during a short distance (10cm).

Is it possible to move a stepper without waiting the end of the move ? maybe just wait a event (the filament detector that detect or not the filament) ?

KevinOConnor, 2020-03-15:
This looks like a dup of #2327 .

-Kevin

kakou-fr, 2020-03-15:
yes i show this issue, but could it work with the extruder ? the SYNC patch looks like it only work for manual_stepper

kakou-fr, 2020-03-17:
I am trying to add the support to sync/desync the extruder with a manual_stepper .
It work for some sync/desync move, but It always crash with this kind of error : 

```
stepcompress o=1 i=0 c=184 a=0: Invalid sequence
```
the current patch :
```diff
diff --git a/klippy/extras/manual_stepper.py b/klippy/extras/manual_stepper.py
index 743c6e23..6f29393e 100644
--- a/klippy/extras/manual_stepper.py
+++ b/klippy/extras/manual_stepper.py
@@ -37,6 +37,8 @@ class ManualStepper:
         self.gcode.register_mux_command('MANUAL_STEPPER', "STEPPER",
                                         stepper_name, self.cmd_MANUAL_STEPPER,
                                         desc=self.cmd_MANUAL_STEPPER_help)
+        # sync state
+        self.sync_state = 0
     def sync_print_time(self):
         toolhead = self.printer.lookup_object('toolhead')
         print_time = toolhead.get_last_move_time()
@@ -74,6 +76,18 @@ class ManualStepper:
         toolhead = self.printer.lookup_object('toolhead')
         toolhead.note_kinematic_activity(self.next_cmd_time)
         self.sync_print_time()
+    def set_extruderstepper(self,enable,extruder):
+        self.sync_print_time()
+        toolhead = self.printer.lookup_object('toolhead')
+        if enable and not self.sync_state:
+            self.rail.set_trapq(extruder.get_trapq())
+            toolhead.register_step_generator(self.rail.generate_steps)
+            self.sync_state = 1
+        elif self.sync_state:
+            self.rail.set_trapq(self.trapq)
+            toolhead.unregister_step_generator(self.rail.generate_steps)
+            self.sync_state = 0
+        self.sync_print_time()
     def do_homing_move(self, movepos, speed, accel, triggered, check_trigger):
         if not self.can_home:
             raise self.gcode.error("No endstop for this manual stepper")
@@ -115,6 +129,10 @@ class ManualStepper:
         if 'SET_POSITION' in params:
             setpos = self.gcode.get_float('SET_POSITION', params)
             self.do_set_position(setpos)
+        if 'SYNC_WITH_EXTRUDER' in params:
+            extruder_name = self.gcode.get_str('EXTRUDER', params, default='extruder')
+            extruder = self.printer.lookup_object(extruder_name)
+            self.set_extruderstepper(self.gcode.get_int('SYNC_WITH_EXTRUDER', params),extruder)
         homing_move = self.gcode.get_int('STOP_ON_ENDSTOP', params, 0)
         speed = self.gcode.get_float('SPEED', params, self.velocity, above=0.)
         accel = self.gcode.get_float('ACCEL', params, self.accel, minval=0.)
diff --git a/klippy/toolhead.py b/klippy/toolhead.py
index 8ce0935c..6fbdfbc3 100644
--- a/klippy/toolhead.py
+++ b/klippy/toolhead.py
@@ -506,6 +506,8 @@ class ToolHead:
         return self.trapq
     def register_step_generator(self, handler):
         self.step_generators.append(handler)
+    def unregister_step_generator(self, handler):
+        self.step_generators.remove(handler)
     def note_step_generation_scan_time(self, delay, old_delay=0.):
         self.flush_step_generation()
         cur_delay = self.kin_flush_delay
```

kakou-fr, 2020-03-21:
ok it work now with an alternative solution based on your previous patch in the issue, I posted this PR : #2636

KevinOConnor, 2020-03-25:
FYI, PR #2636 was merged (commit 6be552f).

-Kevin

kakou-fr, 2020-03-28:
closing now

closed 2020-03-28 by kakou-fr