Klipper3d/klipper #2618: Wax on wax off, can I?

mfp20, 2020-03-14:
tl;dr
I'm offering myself to cleanup/tidyup the code, coherently renaming some labels.

---

I'm developing [a custom module](https://github.com/KevinOConnor/klipper/issues/2007) and I find a bit confusing the (absence of) naming convention and other small cosmetic details up and there. It makes my code digging more complicated than needed. Don't get me wrong, I think it is the normal result of incremental developing, often by multiple developers. Nothing too bad.

Just to make you an example, here is the printer.objects resulting from generic-simulavr.cfg (plus a couple of manual_steppers at the end).

```
INFO:root: ('gcode', <gcode.GCodeParser instance at 0x701b53a26960>)
INFO:root: ('configfile', <configfile.PrinterConfig instance at 0x701b53a337d0>)
INFO:root: ('pins', <pins.PrinterPins instance at 0x701b53a33a50>)
INFO:root: ('heater', <heater.PrinterHeaters instance at 0x701b53a33500>)
INFO:root: ('mcu', <mcu.MCU instance at 0x701b539bed20>)
INFO:root: ('thermistor', None)
INFO:root: ('adc_temperature', None)
INFO:root: ('spi_temperature', None)
INFO:root: ('bme280', None)
INFO:root: ('query_adc', <extras.query_adc.QueryADC instance at 0x701b53755870>)
INFO:root: ('verify_heater heater_bed', <extras.verify_heater.HeaterCheck instance at 0x701b53755f50>)
INFO:root: ('pid_calibrate', <extras.pid_calibrate.PIDCalibrate instance at 0x701b53762320>)
INFO:root: ('heater_bed', <extras.heater_bed.PrinterHeaterBed instance at 0x701b539c1dc0>)
INFO:root: ('fan', <extras.fan.PrinterFan instance at 0x701b53762820>)
INFO:root: ('stepper_enable', <extras.stepper_enable.PrinterStepperEnable instance at 0x701b53770e10>)
INFO:root: ('force_move', <extras.force_move.ForceMove instance at 0x701b537760a0>)
INFO:root: ('query_endstops', <extras.query_endstops.QueryEndstops instance at 0x701b537766e0>)
INFO:root: ('manual_stepper y1', <extras.manual_stepper.ManualStepper instance at 0x701b53770230>)
INFO:root: 
('manual_stepper z1', <extras.manual_stepper.ManualStepper instance at 0x701b53776af0>)
```

As you can see:
- the mandatory steppers (stepper_{x,y,z}) aren't in the OrderedDict, and more in general the mandatory items (ex: mcu) aren't named (ie: it is named just "mcu").
- heating elements, as well as sensors, don't have coherent names, ex: bme280, and thermistor could be named "sense_bme280" and "sense_thermistor", then ordered by name in the OrderedDict. By adding something like "base_" prefix to the general items (ex: gcode, configfile, pins), the end result would be much more comfortable to read.
- ...

Any new developer, or simply a user that would like to customize the code to match his needs, would probably find much easier to find a bit of wax on it when reading the code.
Moreover, in my case I need to have the steppers pointers as I need to swap the {y,z} steppers on tool change, in order to use a different "crane" on my multi-arm printer. So, some changes could be functional to people writing some extra module. One of my development rules was "do not touch pre-existing code", but without those small cosmetic changes it may not be possible to enforce it.
So, before spending time to change the code in my own repo, I'd like to ask [Sensei Kevin](https://www.youtube.com/watch?v=fULNUr0rvEc) if I can do it, and find it pulled upstream.
To be more exact:
- I do not touch ANY logic.
- I do not move ANY code across files.
- ANY functional change (ex: adding the steppers to the OrderedDict) will be notified and included in a single-task set of patches/commit for independent evaluation.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4333293/klippy.log)

KevinOConnor, 2020-03-15:
> I'm offering myself to cleanup/tidyup the code, coherently renaming some labels.

My high-level recommendation would be to start contributing via adding new, generally useful, functionality to Klipper.  I fear that starting contributions by producing "clean up patches" tends to lead to "bike shedding".  If after 6+ months of regular contributions you (or any other developer) wishes to also look at structural changes then that is fine.

> the mandatory steppers (stepper_{x,y,z}) aren't in the OrderedDict

FYI, the individual steppers are owned by their specific kinematic class and thus it is not generally desirable to interact with them directly.  It is possible to get at the steppers as a whole (via the toolhead.get_kinematics().get_steppers() method), but it's probably not a good idea for external code to make assumptions on the layout of the individual kinematic code.

> heating elements, as well as sensors, don't have coherent names, 

FYI, those names match the config section names, which match the module names in klippy/extras/ , which match the user facing documentation in config/example-extras.cfg .

-Kevin

mfp20, 2020-03-16:
@KevinOConnor thanks for the tips! As soon as I handed on the code, I had to shift a bit away from the rules I set to myself :(
I mean, I don't think this can be merged any time soon. I didn't change anything in the core functions (time management, kinematics, comms); and I'm doing anything I can to avoid it; it is too stylish to be changed. But I added a layer between configfile.py and klippy.py.
Basically I wrote a "composer.py" for added flexibility, then called the composer constructor in klippy.Printer._read_config(). It takes elements in the modified configuration file and composes the printer. You can see the "printer tree" in the log.

Given the following printer.cfg

```
#
# SIMPLE PARTS
#

# MCUs
[mcu default]
serial: /tmp/pseudoserial
baud: 250000
pin_map: arduino

# sensors
[sensor min_x]
pin: ^ar1
type: endstop

[sensor max_x]
pin: ^ar2
type: endstop

[sensor min_y]
pin: ^ar3
type: endstop

[sensor max_y]
pin: ^ar4
type: endstop

[sensor min_z]
pin: ^ar5
type: endstop

[sensor max_z]
pin: ^ar6
type: endstop

[sensor h0]
pin: analog0
type: EPCOS 100K B57560G104F

[sensor h1]
pin: analog1
type: EPCOS 100K B57560G104F

[sensor bed]
pin: analog2
type: EPCOS 100K B57560G104F

[sensor tbox]
pin: analog3
type: EPCOS 100K B57560G104F

# steppers
[stepper x]
pin_step: ar7
pin_dir: ar8
pin_enable: ar9
step_distance: .0225

[stepper y]
pin_step: ar10
pin_dir: ar11
pin_enable: ar12
step_distance: .0225
[... 300 lines cut ...]
```
class PrinterPartMap:
    def __init__(self,config):
        self.printer = config.get_printer()
    def register_heater():
        config.get_printer().add_object('heater', PrinterHeaters(config))
    def register_mcu(self, config, name):
        reactor = self.printer.get_reactor()
        mainsync = clocksync.ClockSync(reactor)
        self.printer.add_object('mcu', mcu.MCU(config.getsection(name), mainsync))
        for s in config.get_prefix_sections('mcu '):
            self.printer.add_object(s.section, mcu.MCU(s, clocksync.SecondarySync(reactor, mainsync)))
    def register_extra(self):
        self.try_load_module(config, m.get_name())
    def show_objects(self):
        logging.info("Configured objects:")
        for o in self.printer.objects:
            logging.info("  - %s", o)
```

The composer isn't added to the printer objects, so that memory is freed right after init. Init times could be 1-2secs more. But the added flexibility is huge, the config file is more coherent with the hw, and having a visual representation helps. I simply can't evaluate the performance because:
- I'm currently using a threadripper to run it :)
- I'm not very fluent with python; functional indenting is cool, but I feel uncomfortable without var types and pointers. In any case, I'm avoiding any external lib, and I'm replicating the existing coding style.

Let's see if I can produce something good; maybe can be merged in a -experimental branch, and become -next one day.

mfp20, 2020-03-22:
Just a small update for you to not close the issue. As you are very effective in maintaining the project online...

I'm in the middle of removing the "Printer.objects" because the hardware abstraction layer is complete, so I'm adapting kinematics, stepper and toolhead.

Simple printer:

```
user@user-klipper-devel:~/next$ cat simple.printer.cfg 
# Example config for a simple cartesian printer with 1 extruder and 1 heat bed

#
# SIMPLE PARTS
#

# MCUs
[mcu main]
serial: /tmp/pseudoserial
baud: 250000
pin_map: arduino
restart_method: None

# sensors:
# - type := {endstop, temp, humidity}
# - type:temp, supported models: "EPCOS 100K B57560G104F"
[sensor x1min]
type: endstop
pin: ^ar1

[sensor y1min]
type: endstop
pin: ^ar3

[sensor z1min]
type: endstop
pin: ^ar5

[sensor hblock1]
type: thermometer
model: EPCOS 100K B57560G104F
pin: arm1:analog0

[sensor hbed1]
type: thermometer
model: EPCOS 100K B57560G104F
pin: analog2

# steppers
[stepper x1]
pin_step: ar7
pin_dir: ar8
pin_enable: ar9
step_distance: .0225

[stepper y1]
pin_step: arm1:ar10
pin_dir: arm1:ar11
pin_enable: arm1:ar12
step_distance: .0225

[stepper z1]
pin_step: arm1:ar13
pin_dir: arm1:ar14
pin_enable: arm1:ar15
step_distance: .005

[stepper e1]
pin_step: arm1:ar16
pin_dir: arm1:ar17
pin_enable: arm1:ar18
step_distance: .004242
[... 1031 lines cut ...]
INFO:root:						* HEATER HBLOCK4 TODO heater
INFO:root:							- pin: arm2:ar23
INFO:root:						---
INFO:root:						* SENSOR HBLOCK4 TODO thermometer
INFO:root:							- type: thermometer
INFO:root:							- model: EPCOS 100K B57560G104F
INFO:root:							- pin: arm2:analog1
INFO:root:					---
INFO:root:					* NOZZLE N4 TODO
INFO:root:						- diameter: 0.400
INFO:root:				---
INFO:root:				* FAN COLDEND2 
INFO:root:					- pin: arm2:ar25
INFO:root:				---
INFO:root:				* FAN EXTRUSION2 
INFO:root:					- pin: arm2:ar26
INFO:root:OBJECTS:
INFO:root:  - base_gcode
INFO:root:  - base_configfile
INFO:root:  - multicartesian lowtemp
INFO:root:  - multicartesian midtemp
INFO:root:  - mcu
INFO:root:Starting serial connect
WARNING:root:Unable to open port: [Errno 2] could not open port /tmp/pseudoserial: [Errno 2] No such file or directory: '/tmp/pseudoserial'
```

mfp20, 2020-03-24:
Got rid of printer.objects, and the kinematic-toolhead-extruder-steppers mangle is cleared. Almost ready for the maiden run without the "-v" flag... 

```
user@user-klipper-devel:~/next$ ./test
INFO:root:Starting Klippy...
DEBUG:root:Error getting git version: fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).

WARNING:root:No log file specified! Severe timing issues may result!
INFO:root:Start printer at Tue Mar 24 19:28:18 2020 (1585074498.8 112994.4)
DEBUG:root:- Reading config file.
DEBUG:root:- Composing Printer Tree:
DEBUG:root:     - Reading parts:
DEBUG:root:         - mcu group:
DEBUG:root:             - mcu main
DEBUG:root:             - mcu arm1
DEBUG:root:             - mcu arm2
DEBUG:root:         - sensor group:
DEBUG:root:             - sensor x1min
DEBUG:root:             - sensor x1max
DEBUG:root:             - sensor y1min
DEBUG:root:             - sensor y1max
DEBUG:root:             - sensor z1min
DEBUG:root:             - sensor z1max
DEBUG:root:             - sensor y2min
DEBUG:root:             - sensor y2max
DEBUG:root:             - sensor z2min
DEBUG:root:             - sensor z2max
DEBUG:root:             - sensor hblock1
DEBUG:root:             - sensor hblock2
DEBUG:root:             - sensor hblock3
DEBUG:root:             - sensor hblock4
DEBUG:root:             - sensor hbed1
DEBUG:root:             - sensor tbox
DEBUG:root:         - stepper group:
DEBUG:root:             - stepper x11
DEBUG:root:             - stepper x12
DEBUG:root:             - stepper y1
DEBUG:root:             - stepper z1
DEBUG:root:             - stepper y2
DEBUG:root:             - stepper z21
DEBUG:root:             - stepper z22
DEBUG:root:             - stepper e1
DEBUG:root:             - stepper e2
DEBUG:root:             - stepper e3
DEBUG:root:             - stepper e4
DEBUG:root:         - heater group:
DEBUG:root:             - heater hblock1
DEBUG:root:             - heater hblock2
DEBUG:root:             - heater hblock3
DEBUG:root:             - heater hblock4
DEBUG:root:             - heater hbed1
DEBUG:root:         - nozzle group:
DEBUG:root:             - nozzle n1
DEBUG:root:             - nozzle n2
DEBUG:root:             - nozzle n3
DEBUG:root:             - nozzle n4
DEBUG:root:         - fan group:
DEBUG:root:             - fan coldend1
DEBUG:root:             - fan extrusion1
DEBUG:root:             - fan coldend2
DEBUG:root:             - fan extrusion2
DEBUG:root:     - Reading extra parts: (if any)
DEBUG:root:         - psu main OK
DEBUG:root:         - multicartesian lowtemp OK
DEBUG:root:         - multicartesian midtemp OK
WARNING:root:/home/user/next/klippy/extras/myfake.py DOESN'T EXIST
DEBUG:root:     - Composing complex parts:
DEBUG:root:         - rail group:
DEBUG:root:             - rail bed1
[... 493 lines cut ...]
DEBUG:root:     - cart bed2: <composites.Cart instance at 0x7189ce069460>
DEBUG:root:     - cart midtemp: <composites.Cart instance at 0x7189ce069410>
DEBUG:root:     - tool e3: <composites.Extruder instance at 0x7189ce0693c0>
DEBUG:root:     - stepper e3: <parts.stepper.Stepper instance at 0x7189ce069190>
INFO:root:TODO exec: node.object.configure(hal, node)
DEBUG:root:     - tool hblock3: <composites.Tcontrol instance at 0x7189ce069140>
INFO:root:TODO exec: node.object.configure(hal, node)
DEBUG:root:     - heater hblock3: <parts.heater.Heater instance at 0x7189ce0691e0>
DEBUG:root:     - sensor hblock3: <parts.sensor.Thermometer instance at 0x7189ce0690f0>
DEBUG:root:     - nozzle n3: <parts.nozzle.Nozzle instance at 0x7189ce0690a0>
DEBUG:root:     - tool e4: <composites.Extruder instance at 0x7189ce069050>
DEBUG:root:     - stepper e4: <parts.stepper.Stepper instance at 0x7189ce05daf0>
INFO:root:TODO exec: node.object.configure(hal, node)
DEBUG:root:     - tool hblock4: <composites.Tcontrol instance at 0x7189ce05df00>
INFO:root:TODO exec: node.object.configure(hal, node)
DEBUG:root:     - heater hblock4: <parts.heater.Heater instance at 0x7189ce05da50>
DEBUG:root:     - sensor hblock4: <parts.sensor.Thermometer instance at 0x7189ce05de10>
DEBUG:root:     - nozzle n4: <parts.nozzle.Nozzle instance at 0x7189ce05ddc0>
DEBUG:root:     - fan coldend2: <parts.fan.Fan instance at 0x7189ce05de60>
DEBUG:root:     - fan extrusion2: <parts.fan.Fan instance at 0x7189ce05d780>
INFO:root:Starting serial connect
WARNING:root:Unable to open port: [Errno 2] could not open port /tmp/pseudoserial: [Errno 2] No such file or directory: '/tmp/pseudoserial'
^C

```

mfp20, 2020-03-30:
Still working...

```
user@user-klipper-devel:~/next$ ./test 
INFO:root:Starting Klippy...
DEBUG:root:Error getting git version: fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).

WARNING:root:No log file specified! Severe timing issues may result!
INFO:root:Init printer at Mon Mar 30 04:49:07 2020 (1585536547.5 575027.1)
INFO:root:- Reading config file.
INFO:root:- Composing tree.
INFO:root:- Loading printer objects.
INFO:root:* Init complete.
DEBUG:root:		---
		|  * PRINTER          | <printer.Main instance at 0x7a80232b00f0>
			(event) klippy:ready                  <bound method Parser._handle_ready of <gcode.Parser instance at 0x7a80224c7e60>>
			(event) klippy:shutdown               <bound method MCU._shutdown of <controller.MCU instance at 0x7a8022231af0>>
			(event) klippy:disconnect             <bound method MCU._disconnect of <controller.MCU instance at 0x7a8022231af0>>
			(event) extruder:activate_extruder    <bound method Parser._handle_activate_extruder of <gcode.Parser instance at 0x7a80224c7e60>>
			(event) gcode:request_restart         <bound method Controller.turn_off_all_heaters of <thermometer.Controller instance at 0x7a80222330a0>>
			(event) klippy:connect                <bound method MCU._connect of <controller.MCU instance at 0x7a8022231af0>>
			(event) klippy:mcu_identify           <bound method MCU._mcu_identify of <controller.MCU instance at 0x7a8022231af0>>
			---
			|  * REACTOR          | <reactor.PollReactor instance at 0x7a80232b03c0>
			---
			|  * GCODE            | <gcode.Parser instance at 0x7a80224c7e60>
				(command) ECHO
				(command) FIRMWARE_RESTART
				(command) G0
				(command) G1
				(command) G20
				(command) G21
				(command) G28
				(command) G4
				(command) G90
				(command) G91
				(command) G92
				(command) GET_POSITION
				(command) HELP
				(command) IGNORE
				(command) M105
				(command) M110
				(command) M112
				(command) M114
				(command) M115
				(command) M21
				(command) M220
				(command) M221
				(command) M400
				(command) M82
				(command) M83
				(command) RESTART
				(command) RESTORE_GCODE_STATE
				(command) SAVE_GCODE_STATE
				(command) SET_GCODE_OFFSET
				(command) STATUS
				(command) TURN_OFF_HEATERS
			---
			|  * CONTROLLER       | <controller.Object instance at 0x7a8022233c30>
				---
				|  * CLOCK            | <clock.Primary instance at 0x7a80222339b0>
				---
				|  * THERMOMETER      | <thermometer.Controller instance at 0x7a80222330a0>
				---
				|  * MCU MAIN         | <controller.MCU instance at 0x7a80222337d0>
					serial        : /tmp/pseudoserial
					baud          : 250000
					pin_map       : arduino
					restart_method: None
[... 452 lines cut ...]
							pid_kd  : 114
							min_temp: 0
							max_temp: 210
							---
							|  * SENSOR HBLOCK4   | <parts.sensor_thermometer.Object instance at 0x7a80221fe190>
								type : thermometer
								model: EPCOS 100K B57560G104F
								pin  : arm1:a3
							---
							|  * HEATER HBLOCK4   | <parts.heater.Heater instance at 0x7a80221fe410>
								type: resistor
								pin : arm1:d16
						---
						|  * NOZZLE N4        | <parts.nozzle.Object instance at 0x7a80221fe0a0>
							diameter: 0.400
					---
					|  * COOLER EXTRUSION2  | <parts.cooler_fan.Object instance at 0x7a80221fe370>
						type: fan
						pin : arm1:d20

INFO:root:Starting serial connect
DEBUG:root:Starting stk500v2 leave programmer sequence
DEBUG:root:Got '' from stk500v2
^C
```

mfp20, 2020-04-01:
Almost there. Need to rework the thermal part, and then I publish the code for you to evaluate inclusion.

```
user@user-klipper-devel:~/next$ ./test 
INFO:root:Starting Klippy...
DEBUG:root:Error getting git version: fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).

WARNING:root:No log file specified! Severe timing issues may result!
INFO:root:Init printer at Wed Apr  1 03:59:23 2020 (1585706363.1 110014.4)
INFO:root:- Reading config file.
INFO:root:- Composing printer tree.
WARNING:root:/home/user/next/klippy/extras/myfake.py doesn't exist.
INFO:root:- Loading printer objects.
INFO:root:* Init complete.
DEBUG:root:		---
		|  * PRINTER          | <printer.Main instance at 0x746caea0e9b0>
			(event) extruder:activate_extruder    <bound method Parser._handle_activate_extruder of <gcode.Parser instance at 0x746cadc99140>>
			(event) gcode:request_restart         <bound method Object.turn_off_all_heaters of <temperature.Object instance at 0x746cad9204b0>>
			(event) klippy:connect                <bound method MCU._connect of <controller.MCU instance at 0x746cad91cd70>>
			(event) klippy:disconnect             <bound method MCU._disconnect of <controller.MCU instance at 0x746cad91cd70>>
			(event) klippy:mcu_identify           <bound method MCU._mcu_identify of <controller.MCU instance at 0x746cad91cd70>>
			(event) klippy:ready                  <bound method Parser._handle_ready of <gcode.Parser instance at 0x746cadc99140>>
			(event) klippy:shutdown               <bound method ToolHead._handle_shutdown of <instrument.ToolHead instance at 0x746cad91c4b0>>
			(event) stepper_enable:motor_off      <bound method Robot._motor_off of <kinematics.cartesian.Robot instance at 0x746cad976e60>>
			---
			|  * REACTOR          | <reactor.Poll instance at 0x746caea0ef00>
			---
			|  * GCODE            | <gcode.Parser instance at 0x746cadc99140>
				(command) ECHO
				(command) G0
				(command) G1
				(command) G20
				(command) G21
				(command) G28
				(command) G4
				(command) G90
				(command) G91
				(command) G92
				(command) GCODE_RESTORE_STATE
				(command) GCODE_SAVE_STATE
				(command) GCODE_SET_OFFSET
				(command) GET_POSITION
				(command) HELP
				(command) IGNORE
				(command) M105
				(command) M110
				(command) M112
				(command) M114
				(command) M115
				(command) M204
				(command) M21
				(command) M220
				(command) M221
				(command) M400
				(command) M82
				(command) M83
				(command) RESTART
				(command) RESTART_FIRMWARE
				(command) SET_VELOCITY_LIMIT
				(command) SHOW_PINS_ACTIVE
				(command) SHOW_PINS_ALL
				(command) SHOW_PRINTER
				(command) SHOW_STATUS
				(command) TURN_OFF_HEATERS
			---
			|  * CONTROLLER       | <controller.Object instance at 0x746cad920820>
				- board arm1:	(pin)(pullup)(invert)
							none
				- board arm2:	(pin)(pullup)(invert)
[... 518 lines cut ...]
DEBUG:root:Ignoring clock sample 110020.839: freq=2016962 diff=540330 stddev=19963.911
DEBUG:root:Ignoring clock sample 110021.105: freq=2287776 diff=1394307 stddev=22790.930
DEBUG:root:Ignoring clock sample 110021.823: freq=2016962 diff=1319899 stddev=19963.911
DEBUG:root:new minimum rtt 110021.981: hrtt=0.005345 freq=20000000
INFO:root:Resetting prediction variance 110021.981: freq=20000000 diff=-300850 stddev=20000.000
DEBUG:root:new minimum rtt 110022.047: hrtt=0.004565 freq=1292573
DEBUG:root:Ignoring clock sample 110022.090: freq=2287776 diff=1738069 stddev=22790.930
DEBUG:root:new minimum rtt 110022.177: hrtt=0.002978 freq=2308557
INFO:root:Loaded MCU 'arm2' 80 commands (v0.8.0-367-g0ba83c5b-20200314_085106-user-klipper-devel / gcc: (GCC) 5.4.0 binutils: (GNU Binutils) 2.26.20160125)
MCU 'arm2' config: RECEIVE_WINDOW=192 BUS_PINS_spi=PB6,PB5,PB7 SERIAL_BAUD=250000 STEP_DELAY=-1 ADC_MAX=1023 PWM_MAX=255 RESERVE_PINS_serial=PD0,PD1 BUS_PINS_twi=PC0,PC1 MCU=atmega644p STATS_SUMSQ_BASE=256 CLOCK_FREQ=20000000
INFO:root:Sending MCU 'main' printer configuration...
INFO:root:Configured MCU 'main' (283 moves)
INFO:root:Sending MCU 'arm1' printer configuration...
INFO:root:Configured MCU 'arm1' (1024 moves)
INFO:root:Sending MCU 'arm2' printer configuration...
INFO:root:Configured MCU 'arm2' (1024 moves)
DEBUG:root:Ignoring clock sample 110022.808: freq=2016962 diff=1835130 stddev=19963.911
DEBUG:root:Ignoring clock sample 110023.074: freq=2287776 diff=2032937 stddev=22790.930
DEBUG:root:Ignoring clock sample 110023.792: freq=2016962 diff=2622075 stddev=19963.911
DEBUG:root:Ignoring clock sample 110024.059: freq=2287776 diff=2233413 stddev=22790.930
DEBUG:root:Ignoring clock sample 110024.777: freq=2016962 diff=3209202 stddev=19963.911
DEBUG:root:Ignoring clock sample 110025.043: freq=2287776 diff=2577597 stddev=22790.930
INFO:root:Resetting prediction variance 110025.443: freq=2693346 diff=-298785 stddev=28085.893
DEBUG:root:new minimum rtt 110025.762: hrtt=0.002708 freq=2016962
```

mfp20, 2020-04-03:
Had to rework gcode.py too, in order to have multiple toolheads. Now each toolhead has its own kinematics and gcode.
All toolheads share the same printer facilities (hal, reactor, commander, controller, timing, temperature). Commander is some of the code from gcode.py; it reads commands and dispatch them to other commanders for their own toolheads, for parameters manipulation (or command abortion), before executing the command handler. It should be enough to allow the legacy movement logic to execute properly.

At this point I need some way to push gcodes to my simulavr(s) and see what happens...

```
user@user-klipper-devel:~/next$ ./test
INFO:root:Starting Klippy...
DEBUG:root:Error getting git version: fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).

WARNING:root:No log file specified! Severe timing issues may result!
INFO:root:Init printer at Fri Apr  3 19:14:47 2020 (1585934087.8 337733.2)
INFO:root:- Reading config file.
INFO:root:- Composing printer tree.
WARNING:root:/home/user/next/klippy/extras/myfake.py doesn't exist.
INFO:root:- Loading printer objects.
DEBUG:root:	commander
DEBUG:root:	controller
DEBUG:root:	timing
DEBUG:root:	temperature
DEBUG:root:	mcu main
DEBUG:root:	mcu arm1
DEBUG:root:	mcu arm2
DEBUG:root:	toolhead dummy
WARNING:root:DummyToolHead:__init__()
DEBUG:root:	kinematic dummy
DEBUG:root:	toolhead lowtemp
DEBUG:root:	kinematic lowtemp
DEBUG:root:	rail bed1
DEBUG:root:	stepper x11
DEBUG:root:	sensor x1min
DEBUG:root:	sensor x1max
DEBUG:root:	rail head1
DEBUG:root:	stepper y1
DEBUG:root:	sensor y1min
DEBUG:root:	sensor y1max
DEBUG:root:	rail arm1
DEBUG:root:	stepper z1
DEBUG:root:	sensor z1min
DEBUG:root:	sensor z1max
DEBUG:root:	cart bed1
DEBUG:root:	tool hbed1
DEBUG:root:	sensor hbed1
DEBUG:root:	heater hbed1
DEBUG:root:	cart lowtemp
DEBUG:root:	tool e1
DEBUG:root:	stepper e1
DEBUG:root:	tool hblock1
DEBUG:root:	sensor hblock1
DEBUG:root:	heater hblock1
DEBUG:root:	cooler coldend1
DEBUG:root:	nozzle n1
DEBUG:root:	tool e2
DEBUG:root:	stepper e2
DEBUG:root:	tool hblock2
DEBUG:root:	sensor hblock2
DEBUG:root:	heater hblock2
DEBUG:root:	nozzle n2
DEBUG:root:	cooler extrusion1
DEBUG:root:	toolhead midtemp
DEBUG:root:	kinematic midtemp
DEBUG:root:	rail bed2
DEBUG:root:	stepper x12
DEBUG:root:	rail head2
DEBUG:root:	stepper y2
DEBUG:root:	sensor y2min
DEBUG:root:	sensor y2max
DEBUG:root:	rail arm2
DEBUG:root:	stepper z21
[... 944 lines cut ...]
				---
				|  * SENSOR BOXBOTTOM  | <parts.thermometer.Object instance at 0x739a4f2b4870>
					type : thermometer
					model: Honeywell 100K 135-104LAG-J01
					pin  : arm1:a7
				---
				|  * COOLER BOX       | <parts.fan.Object instance at 0x739a4f2b4e60>
					type: fan
					pin : arm1:d21

		---
		|  * SPARES          

DEBUG:root:Ignoring clock sample 337741.328: freq=2341794 diff=813408 stddev=26061.659
DEBUG:root:Ignoring clock sample 337741.989: freq=2076659 diff=546228 stddev=23137.309
DEBUG:root:Ignoring clock sample 337742.313: freq=2341794 diff=776537 stddev=26061.659
DEBUG:root:new minimum rtt 337742.974: hrtt=0.002871 freq=2076659
DEBUG:root:Ignoring clock sample 337742.974: freq=2076659 diff=1162439 stddev=23137.309
DEBUG:root:Ignoring clock sample 337743.297: freq=2341794 diff=1243725 stddev=26061.659
DEBUG:root:Ignoring clock sample 337743.658: freq=2423972 diff=520314 stddev=52259.146
DEBUG:root:new minimum rtt 337743.959: hrtt=0.002368 freq=2076659
DEBUG:root:Ignoring clock sample 337743.959: freq=2076659 diff=1978680 stddev=23137.309
DEBUG:root:Ignoring clock sample 337744.282: freq=2341794 diff=1630938 stddev=26061.659
DEBUG:root:Ignoring clock sample 337744.643: freq=2423972 diff=586674 stddev=52259.146
```

KevinOConnor, 2020-04-24:
I'm going to close this as it looks like the original question was answered and the conversation seems to have moved to #2683 .

-Kevin

closed 2020-04-24 by KevinOConnor