Klipper3d/klipper #2619: Layer Missing/skipped layers and odd layer colours in a Creality CR10s-Pro + E3V6 + Bondtech

netonjm, 2020-03-15:
Hello! First of all thank @KevinOConnor and the whole team behind this open source product. I am completely in love with him and I hope to contribute a little in the future.

After a wonderful learning path, I already have my Creality CR10S Pro working with Klipper, it has been a bit more complex than expected since I had no specific configuration for my Creality board model 2.4, for which there was not much documentation but looks like very similar to the CR10/S, so I have used this base model, with some change of pins and it seems to work.

Additionally did some changes to the printer to make it a direct unit... added a BMG (bondtech) and an E3V6 with a 5015 fan.

But before trying it on Klipper I wanted to make sure there was no mechanical issues, so I tried it on Marlin first getting very good results:

.[image]

[image]

Unfortunately, when trying to test my configuration using the same 3d printing software (no matter what I use I get same behaviour/pattern) with the same configuration, the results are not as expected, making a constant and strange printing patterns  with some color changes (temperature fluctuation or loss of steps?)

[image]

[image]

[image]

I don't know what else to try, I would be very grateful if someone could help me.

Greetings.

Log:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4333623/klippy.log)

BlackStump, 2020-03-15:
try upgrading to the latest version 
https://www.klipper3d.org/FAQ.html#how-do-i-upgrade-to-the-latest-software

netonjm, 2020-03-15:
I updated to most recent hash in master branch following the guide:

[image]

[klippy.log](https://github.com/KevinOConnor/klipper/files/4333789/klippy.log)

Hywelmartin, 2020-03-15:
[heater_bed]
heater_pin = ar8
sensor_type = EPCOS 100K B57560G104F
sensor_pin = analog14
control = watermark
min_temp = 0
max_temp = 130

---->

[heater_bed]
heater_pin = ar8
sensor_type = EPCOS 100K B57560G104F
sensor_pin = analog14
control: pid
pid_Kp: 22.2
pid_Ki: 1.08
pid_Kd: 114
min_temp: 0
max_temp: 130

and make a PID tune....

KevinOConnor, 2020-03-15:
As @Hywelmartin indicates - this type of "z banding" is common when using the "watermark" heater - try using the pid heater.

Also, do upgrade to the very latest version (from just a few hours ago) as PR #2597 is important if you are running the tmc2208 in standalone mode.  See the FAQ for upgrade instructions.

-Kevin

netonjm, 2020-03-15:
thank you guys! I just added the heater bed to use pid and now all works like a charm! 

[image]

the first test and it is the best calibration cube i have ever made. Great job!

closed 2020-03-15 by netonjm