Klipper3d/klipper #2622: Endstop z still triggered after retract (ABL probe)

multilaton, 2020-03-16:
Hi, 
I have ender3 and klipper work flawlessly. I include capacitive sensor to my ender3 to gain ABL feature.
I did everything in the documentation. İt works at first but now it bheave strange. When ı want to home all axes it home first x and y (normally) but then the z axis sensor being triggered. And with triggering z axes probe it didnt make homing for z axes. And give me trouble as shown in title "Endstop z still triggered after retract".
When I turn the potentiometer of sensor and reduce the range of capacitiveness it works.But again I have this trouble.
Could anyone help me ?
Btw Klipper is just awesome Thanks KevinOConnor and All community members.

KevinOConnor, 2020-03-29:
Unfortunately, it doesn't appear anyone has much assistance to offer.  I, personally, have no experience with capacitive probes.  It does sound like a hardware issue of some kind.

-Kevin

closed 2020-04-08 by KevinOConnor

multilaton, 2020-04-20:
I solved my problem.
There was noise in my probe signal cable.
Then I cover them with noise filter(a thing that found on usb cable, some magnetic shield I guess).
[image]