Klipper3d/klipper #2623: DWC PSU control

shadrincev, 2020-03-17:
Good afternoon. How can I activate the enable / disable button? I write a macro for the M80 and M81 commands that are sent when it is clicked, but an error occurs: Malformed command 'RESPOND type:command MSG=action:poweron'. Or is this also not implemented?

```
[respond]
default_type: echo

[gcode_macro M80]
gcode: RESPOND type:command MSG=action:poweron

# power off
[gcode_macro M81]
gcode: RESPOND type:command MSG=action:poweroff
```
To run gpio did this:
```
cd /home/octo
git clone https://github.com/zhaolei/WiringOP.git -b h3 && cd WiringOP
chmod +x ./build
sudo ./build
cd ..
gpio readall
```
a table with gpio outputs appears.
Create a folder for scripts to enable and disable printers and download the scripts themselves.
```
mkdir power_scripts
cd power_scripts
wget https://raw.githubusercontent.com/Nebari-xx/octoprint_installer/master/poweron.sh
wget https://raw.githubusercontent.com/Nebari-xx/octoprint_installer/master/poweroff.sh
chmod +x poweron.sh
chmod +x poweroff.sh
```
Check the performance: 
Turning on the printer
`./poweron.sh`
Turning off the printer
`./poweroff.sh`
The issue arose due to the use of the DWC interface . It sends commands M80 and M81, but somehow they are not executed. Please tell me what I'm doing wrong.

ld3300, 2020-03-17:
capitalize TYPE:

shadrincev, 2020-03-18:
Thanks. Even with the sign = swears at the type parameter. Without it, the command is executed , but the relay is not activated. Maybe the clipper just can 't send commands to the terminal? And I'm prescribing all this for nothing...

ld3300, 2020-03-18:
It should be a response message to the host that is sending the G-code. In octoprint you need a plug-in to trigger a response to action commands. Last time I tried dwc i gave up because it did not yet seen to have the ability to respond to custom actions, but I am an amateur, so maybe someone with more knowledge can fill in the gaps.

vladbabii, 2020-03-19:
https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md

```
M118 <message>: echo the message prepended with the configured default prefix (or echo: if no prefix is configured).
RESPOND MSG="<message>": echo the message prepended with the configured default prefix (or echo: if no prefix is configured).
RESPOND TYPE=echo MSG="<message>": echo the message prepended with echo: .
RESPOND TYPE=command MSG="<message>": echo the message prepended with // . Octopint can be configured to respond to these messages (e.g. RESPOND TYPE=command MSG=action:pause).
RESPOND TYPE=error MSG="<message>": echo the message prepended with !! .
RESPOND PREFIX=<prefix> MSG="<message>": echo the message prepended with <prefix> (The PREFIX parameter will take priority over the TYPE parameter)
```

not sure where you took "type:command" from but it seems wrong. If it's something from DWC it's not supported by official klipper, you'll need help from the DWC maintainer

closed 2020-03-24 by klipper-gitissuebot