Klipper3d/klipper #2627: Extruder fails to query or maintain temperature at start of print

argillaceous79, 2020-03-17:
Good Day,

I've been struggling with my Maker Select v2.1 (with factory bootloader but without pullup resistors) for a couple of weeks now.  Once I start a print the extruder and the bed heater begin dropping in temperature quickly enough that the chamber never finishes priming.  I've verified the thermistor functions and is accurate using a multimeter thermometer.  I've noticed when watching the temperature on the LCD screen, on the Octoprint readout, and even the Cura readout that the temperature only fluctuates .1 degree after the start of the print.  If I send M105 it takes a minute or two before it retrieves the correct temperature and stops the print due to temperature threshold.  If I don't manually request the temperature it keeps going through the print and never realizes the extruder is cold (and shredding my filament)

What is very perplexing to me is that when not in a print, the extruder heats up and extrudes completely normally, and the temperature is updating continuously like it should.  To further add to my confusion, I was able to tune pressure advance by printing the tower using the instructions provided.  I've checked my start_print code, moved the start code from slicer to klipper macro in the config file, to octoprint, and even tried a different slicer.  I've tried several variations of start _print code.  

Am I just missing something extremely basic, or is there something else going on here?

[gcode_macro START_PRINT]
default_parameter_BED_TEMP: 60
default_parameter_EXTRUDER_TEMP: 210
gcode:
  SET_GCODE_OFFSET Z=0.0; z offset reset
  M140 S{BED_TEMP}; set bed temperature
  G90; absolute coordinates
  G28; home all
  G1 Z0.15 F300; move z close to bed
  M190 S{BED_TEMP}; wait for bed temp
  M109 S{EXTRUDER_TEMP}; wait for extruder temp

[klippy.log](https://github.com/KevinOConnor/klipper/files/4344862/klippy.log)

KevinOConnor, 2020-03-18:
You don't actually have a `[gcode_macro START_PRINT]` defined in your printer.cfg file.  As a result, no G28 was run and an error caused OctoPrint to turn off the heaters.  Double check that you are editing the correct config file and that it got properly saved.

-Kevin

argillaceous79, 2020-03-18:
I'm sorry, I must have left the start and end code in the slicer before I extracted that log.  Here is a log from an attempt I just ran, with the code back in the printer.cfg file.  I also made sure that it shows up in the OctoKlipper printer.cfg, and for good measure checked it against the printer.cfg in the pi's user directory via sftp.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4346960/klippy.log)

ld3300, 2020-03-19:
I have no clue where the problem is coming from, but your hot end will heat faster if you put the 109 before the 190 (since the bed is already on it's way to the temperature.)
I'd be really interested in seeing the full g-code file that you are printing that is causing the failure.

argillaceous79, 2020-03-19:
Thanks for the tip, I made the change.  

But more importantly, I used 2 different slicers (Cura and Slic3r) with the same results.  I don't think its the slicer.  Here is the gcode if you want a look.  I ordered another thermistor just on the off chance this one too is bad.  One thing I think was different this time is that I saw temperature messages at about every 30 seconds or so.  They were reporting a temperature close to the temperature it was supposed to be and not the actual temperature which was already below melting temp by the time it started on the skirt.

[uno_case.txt](https://github.com/KevinOConnor/klipper/files/4352676/uno_case.txt)

argillaceous79, 2020-03-19:
So I still have not isolated the cause of the heat problem, but I switched stl and slicer (Prusa Slicer).  Extruder and bed are both maintaining temperature now.  At least we can be reasonably sure it isn't klipper.

closed 2020-03-19 by argillaceous79

ld3300, 2020-03-20:
If the default min temp is too low for your filament you might want to increase that too, at least it should stop before the filament can no longer flow.