Klipper3d/klipper #2628: [Feature][Brainstorming] Fallback endstop support

Depau, 2020-03-18:
A feature that I would be interested in would be fallback endstop.
That is additional endstops that can be specified for each stepper that, if triggered, will also stop movement.

My use case is this: I'm planning to switch to optical endstops, but I have the TMC2209 with the StallGuard, that can be used as a virtual endstop.

My idea is that the StallGuard feature could be used as a fallback endstop in case, for example, the plastic strip that triggers the optical endstop breaks off.

That wouldn't be an ideal solution in the long term, since using different endstops other than the main ones would screw up all the offsets and, in general, a print wouldn't be feasible.

Also, I know that klipper times out homing pretty quickly if the endstop doesn't get triggered. However, the noise it makes is quite loud and not pleasing, and I have neighbors I'd like to have a good relationship with ;) Also it causes mechanical stress that can be avoided.

I'd be willing to implement it my self and send a pull request if this is a feature you'd be interested in.

My idea would be to add an item to the stepper configurations, like this:

```
[stepper_x]
...
endstop_pin: P1
endstop_fallback_pins: tmc2209_stepper_x:virtual_endstop, P2, P3, ...
```

I'd make it so that in case the main endstop it triggered, the behavior is unchanged. However, when one of the fallback ones is triggered, the firmware shuts down.

Let me know if this feature would be interesting to you. Anybody, if you have any suggestions for a better behavior, also let me know :)

----

Here bot, take this fake log: [klippy.log](https://github.com/KevinOConnor/klipper/files/4351157/klippy.log)

KevinOConnor, 2020-03-21:
There isn't currently support for "run time" endstop detection.  If you just want a safety check, you could declare `gcode_button` config sections for each additional endstop and have the config raise a shutdown on a trigger event.

Separately, it is not currently possible to detect tmc stall events at run-time.  The support for stall detection is different on each tmc driver, and not all of them support stall notification when the driver is configured in its normal printing mode.  I suspect it would be possible to implement some kind of stall detection on some of the drivers if someone was interested in developing that, but I believe there would still be limitations imposed by the hardware.

-Kevin

Depau, 2020-03-21:
Hi Kevin,
I forgot to mention that detail but I was interested in getting that behavior only while homing (in case the regular endstop somehow breaks). While it's printing, Klipper already does its own boundary checking and I'm good with that.

I didn't know about `gcode_button`, I'll test it and see how it works.

schnello, 2020-03-26:
> That is additional endstops that can be specified for each stepper that, if triggered, will also stop movement.

What you describe is nothing else as the "filament runout sensor".
The sensor (endstop) can execute gcode and stop the print if the endstop is triggered.