Klipper3d/klipper #2630: Internal error on command:"G28"

Oranginator, 2020-03-18:
Hi together,

i just wanted to home my printer with G28 after switching from RADDS-Board to SKR 1.1 Pro. Now I'm getting this error and I couldn't figure out myself why i get it.

After printing G28 into the terminal and hiting the enter-key my printer shuts down immediatly. 

Showing the following message:

Internal error on command:"G28"
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

I could verify all endstops as not being triggered and the motors are moving when I use the STEPPER_BUZZ command.

Restarting Firmware, Service, RaspberryPi and totaly switching of the printer weren't changing anything.

Perhaps you guys can help me, thanks in advance.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4351717/klippy.log)

ld3300, 2020-03-19:
in your stepper_y why is your homing_positive_dir = 0?

Oranginator, 2020-03-19:
> 
> 
> in your stepper_y why is your homing_positive_dir = 0?

Thank you. Problem solved.

closed 2020-03-19 by Oranginator