Klipper3d/klipper #2633: can't get klipper to make /tmp/klipper after messing with the code

AmyTheCute, 2020-03-20:
Sadly it doesn't even make a klippy.log now.
I tried adding M204 to the list of gcode commands on gcode.py and made a function cmd_M204 that for now only executed this line: self.toolhead.max_accel = 100
i was trying to understand the code of this firmware and considering im a beginner i'm having a hard time. i thought this was a good challenge/learning experience plus helpful

after I re ran the install-octopi.sh everything went well expect now I can't see a /tmp/klipper/ i've tried restarting it but I can't figure out what's happening.

I obviously tried removing my changes but it didn't help whatsoever, and i"m lost. this is my actual raspberry pi for klipper and I made the dumb mistake of using it for my testing sine windows linux subsystem was being annoying. 

Thanks

AmyTheCute, 2020-03-20:
Solved by letting the printer sit for a while?! strange..... let me re commit my change again....

closed 2020-03-20 by AmyTheCute

ld3300, 2020-03-20:
Hi @AmyTheCute .  Since the klipper code is already a git system I created a branch that I screw around with, and if I make a mistake I can revert back to the master.  It required me to learn a few terminal git commands, but I got there.  You can also revert at any time to the unmodified klipper code by ssh-ing in, changing to the klipper directory, and doing git reset --hard origin/master.  Since the config file isn't in that directory it shouldn't be affected.

AmyTheCute, 2020-03-21:
> Hi @AmyTheCute . Since the klipper code is already a git system I created a branch that I screw around with, and if I make a mistake I can revert back to the master. It required me to learn a few [...]

Thanks! but I should probably understand the freaking code first xD this is soo complicated and I've been reading the code over and all but it's hard to understand for a beginner like me