Klipper3d/klipper #2634: Macro - Lost communication with MCU 'mcu'

boboxx, 2020-03-20:
I have a macro that was used to warm up the host end and bed but since I updated to the lates version I'm getting a "Lost communication" every-time I call the macro.

Klipper Message:
12:15:02 Lost communication with MCU 'mcu'
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown
12:15:02 Lost communication with MCU 'mcu'

The macro I'm calling:
[gcode_macro ENDER_WARMUP]
gcode:
  M140 S60
  M104 S210

ld3300, 2020-03-20:
@boboxx please include your log file, as requested by the submit issue doc.  It will give a lot more information about what is going on.

Is the code exactly as you have posted it, or are there indents in the gcode commands as required? ie:

```
[gcode_macro ENDER_WARMUP]
gcode:
    M140 S60
    M104 S210
```

boboxx, 2020-03-21:
I have reformated my card and reinstalled Klipper and it solved all my problem. Everythis is working now as it was.

closed 2020-03-21 by boboxx