Klipper3d/klipper #2637: T1 command not recognized.

AmyTheCute, 2020-03-21:
For some reason klipper doesn't want to do T1 anymore. when I type the command it tells me that the command is unknown, Strangely for T3 it tells me that no such extruder exists (which is true)

I typed m112 a few times but strangely my log is empty. I even tried deleting it and doing it and it didn't even make a new one. It's very strange. so I will include my config file the only thing I have, and the 

[21 march 2020 klippedy.txt](https://github.com/KevinOConnor/klipper/files/4363339/21.march.2020.klippedy.txt)

[klippy.log](https://github.com/KevinOConnor/klipper/files/4363340/klippy.log) (empty...)

AmyTheCute, 2020-03-21:
huh the log appeared out of nowhere after a few minutes. strange
[klippy.log](https://github.com/KevinOConnor/klipper/files/4363386/klippy.log)
I guess it just takes long

ld3300, 2020-03-21:
Hi Amy,
Go read the config changes file at [Config_Changes.md](https://github.com/KevinOConnor/klipper/blob/master/docs/Config_Changes.md)
Especially the December 10th update entry.

The T0 and T1, (etc) are no longer built in G-Code commands.  This has made it a lot more flexible since you can declare them as gcode macros and have more control over the behavior.  It also requires setting the active extruder as described in the doc.  It has made my tool change behavior much easier to manage.

AmyTheCute, 2020-03-21:
Ooh I didn't notice that! must have gotten an update when I tried to re install klipper! <3 Thanks a ton!
I'm still confused about what I have to do with my offsets to work as it did before? do I just add move=1 and it's done? since I used to move the bed down manually with g91 before setting the offsets because it didn't apply it until the next layer change

closed 2020-03-21 by AmyTheCute

ld3300, 2020-03-22:
The SET_GCODE_OFFSET command should take affect on the next absolute position G1 command that includes that axis, so X and Y should take effect quickly, but I can see how that might mess the Z up.  So adding the MOVE=1 seems it would resolve any concern and get the nozzle where you want it during the switch.  I don't think you should need to use G91 at all unless there is a problem.  The example in [sample_idex.cfg](https://github.com/KevinOConnor/klipper/blob/d3a7d992aa8a2a20dc1923985a6e292374e30f01/config/sample-idex.cfg#L44) doesn't use it (MOVE=1), but my setup is a dual extruder single hotend setup, so the offsets don't apply and I can't contribute to how effective that part of the sample is.