Klipper3d/klipper #2639: Can't delete files on VirtualSD through OctoPrint

mejobloggs, 2020-03-22:
Is this a Klipper thing or an OctoPrint thing?

I click delete in OctoPrint and I get this error:

```
Send: M30 /ce3_m6socketandwrench.gcode
Recv: !! SD write not supported
```
I have clicked the "Refresh SD Files" button in OctoPrint and confirmed the file hasn't deleted.

Also, is it possible to create folders on VirtualSD? That doesn't work for me either
[klippylog.zip](https://github.com/KevinOConnor/klipper/files/4364556/klippylog.zip)

Hywelmartin, 2020-03-22:
remove the "local" file.. and then update

swatti87, 2020-03-22:
Is there a way to "delete all" in local, a macro or something? Stuff tends to pile up in there.

KevinOConnor, 2020-03-23:
The virtual_sdcard does not support uploading, deleting, or modifying of any files.  It just allows printing of files already in a particular system directory.  The typical usage would be to point the virtual_sdcard to OctoPrint's upload directory, and then use OctoPrint to manage the files.  (Delete a file from OctoPrint and it will be removed from the available list.)  When using OctoPrint I find it handy to click on the wrench icon above the Files list - select "only show files stored on sd" when printing, and select "only show files stored locally" when deleting files.  It may be necessary to click on "refresh SD files" after making a change.

-Kevin

mejobloggs, 2020-03-24:
Thanks for the explanation Kevin, I understand now

closed 2020-03-24 by mejobloggs