Klipper3d/klipper #2641: SKR 1.4 + TMC5160 is not working

MeisterPilaf, 2020-03-23:
Hi, 

i cant home or move any motor with my config. I really tested everything. 

The pins on the board are correct. The connection from the motor to the mainboard are correct and working (testet with a multimeter). 

I dont know what i do wrong. I also tryed the command "STEPPER_BUZZ STEPPER=stepper_x" nothing happend. 

[klippy.log](https://github.com/KevinOConnor/klipper/files/4368215/klippy.log)

MeisterPilaf, 2020-03-23:
========== Write-only registers ==========
TPWMTHRS: 00000013 TPWMTHRS=19
COOLCONF: 00000000
GLOBALSCALER: 00000054 GLOBALSCALER=84
IHOLD_IRUN: 00061f0f IHOLD=15 IRUN=31 IHOLDDELAY=6
PWMCONF: c40d001e PWM_OFS=30 pwm_freq=1 pwm_autoscale=1 pwm_autograd=1 PWM_REG=4 PWM_LIM=12
TPOWERDOWN: 0000000a TPOWERDOWN=10
========== Queried registers ==========
GCONF: ffffffff recalibrate=1 faststandstill=1 en_pwm_mode=1 multistep_filt=1 shaft=1(Reverse) diag0_error=1 diag0_otpw=1 diag0_stall=1 diag1_stall=1 diag1_index=1 diag1_onstate=1 diag1_steps_skipped=1 diag0_int_pushpull=1 diag1_poscomp_pushpull=1 small_hysteresis=1 stop_enable=1 direct_mode=1 test_mode=1
CHOPCONF: ffffffff toff=15 hstrt=7 hend=15 fd3=1 disfdcc=1 chm=1 tbl=3 vhighfs=1 vhighchm=1 tpfd=15 MRES=15(0usteps) intpol=1 dedge=1 diss2g=1 diss2vs=1
GSTAT: ffffffff reset=1(reset) drv_err=1(ErrorShutdown!) uv_cp=1(Undervoltage!)
DRV_STATUS: ffffffff SG_RESULT=1023 s2vsa=1 s2vsb=1 stealth=1 fsactive=1 CSACTUAL=255 stallGuard=1 ot=1(OvertempError!) otpw=1(OvertempWarning!) s2ga=1(ShortToGND_A!) s2gb=1(ShortToGND_B!) ola=1(OpenLoad_A!) olb=1(OpenLoad_B!) stst=1
FACTORY_CONF: ffffffff FACTORY_CONF=31
IOIN: ffffffff REFL_STEP=1 REFR_DIR=1 ENCB_DCEN_CFG4=1 ENCA_DCIN_CFG5=1 DRV_ENN=1 ENC_N_DCO_CFG6=1 SD_MODE=1 SWCOMP_IN=1 VERSION=0xff
LOST_STEPS: ffffffff LOST_STEPS=1048575
MSCNT: ffffffff MSCNT=1023
MSCURACT: ffffffff CUR_A=-1 CUR_B=-1
OTP_READ: ffffffff OTP_FCLKTRIM=31 otp_S2_LEVEL=1 otp_BBM=1 otp_TBL=1
PWM_SCALE: ffffffff PWM_SCALE_SUM=255 PWM_SCALE_AUTO=-1
PWM_AUTO: ffffffff PWM_OFS_AUTO=255 PWM_GRAD_AUTO=255
TSTEP: ffffffff TSTEP=1048575

MeisterPilaf, 2020-03-24:
Ok one driver was delivered broken. -.-

KevinOConnor, 2020-04-08:
I'm going to close this as it appears the root cause of the issue was found.

-Kevin

closed 2020-04-08 by KevinOConnor