Klipper3d/klipper #2642: Z homing & BLtouch issue

sedlacekdavid, 2020-03-23:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4368534/klippy.log)

VIDEO here:
https://youtu.be/3UncA8cA1fs

Hi, I'm having strange issues with my BLtouch clone. Everything has been working fine for two months until yesterday. 

It behaves really strange - when I'm Z homing having BLtouch as a probe, it just fails everytime on the second "touch". 

This happens also when I'm bed mesh probing, it seems it just can't see bltouch retraction but in some very specific cases - like on the second touch shown on the video below. The board/firmware is obviously (at least it seems so) getting feedback from the probe and obviously the board can control the probe. So therefore I'm assuming the physical wiring is still intact. I have a scope available so I can check. But on the video it is clear that the board can detect Z touch and stops moving further down.

I've messed with parameters but none of that has fixed the problem ......

sedlacekdavid, 2020-03-23:
One more thing - on my Pi I have sort of automatic GIT Pull. So I'm basically running the latest Master.

sedlacekdavid, 2020-03-23:
Sorry, a big update -> when I've been messing with the config, of course I have changed pin_up_touch_mode_reports_triggered,
so it is of course like:

`pin_up_touch_mode_reports_triggered: False` - it's never worked without this. 

So what really happens now I can obviously correctly Z-home, but when I execute PROBE, it just moves down straight to the board. The only thing that stops movement are soft endstops. 

Video here: https://youtu.be/_jeAcwsykWw

sedlacekdavid, 2020-03-23:
I'm not really sure if there's something with my probe so I should replace it, but everything seems like I can control it and there is also a proper feedback...

sedlacekdavid, 2020-03-23:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4368673/klippy.log)

GarthSnyder, 2020-03-23:
> One more thing - on my Pi I have sort of automatic GIT Pull. So I'm basically running the latest Master.

If you revert to known-good version of Klipper, does the problem still occur?

sedlacekdavid, 2020-03-23:
Guys, it seems everything's fine now. I don't really know what was the initial root cause or if it was just a change from Master. But, I've changed some settings related to retraction distances etc and now it is perfect. Whole bltouch procedure of 7x7 points takes around 3-4 minutes, which is amazing.

https://www.youtube.com/watch?v=CFFPIKMvNF0

@GarthSnyder thank you very much for reviewing this.

closed 2020-03-23 by sedlacekdavid