Klipper3d/klipper #2643: [dual nozzle]:Z-HOP during fct wait_for_temperature

malphodo, 2020-03-24:
Hello all,

My first contribution, thank you for being indulgent if I make a mistake in the method ;)

It is very difficult to use Klipper with a dual nozzle on one carriage like Chimera.

During hotend changes with somes macros (T0,T1) , the temperature rises with the nozzle in contact with the print, which deforms it.

Would it be possible to have a configurable Z-Hop on the heating phases?

Input condition : more than 1 hotend
Variable in .CFG : Heat_z_distance
maybe called by the function wait_for_temperature

what do you think about this ?

BeatSlayer, 2020-03-24:
Raisong the z in the T0/T1 macros and waiting to extruder temp before lowering z to continue printing should work.

malphodo, 2020-03-26:
> Raising the z in the T0/T1 macros and waiting to extruder temp before lowering z to continue printing should work.
> 
> Edit: Typo

Thank you for your suggestion.
I had to return to Marlin for an urgent impression.
But I tried to work on the macro, what do you think about it ? I haven't tested it yet.

```
[gcode_macro T0]
gcode:
    {% if not printer.toolhead.extruder == "extruder" %}
    ACTIVATE_EXTRUDER EXTRUDER=extruder
    SET_GCODE_OFFSET Z=1 MOVE=1 MOVE_SPEED=400              # Adjust z-height
    SET_GCODE_OFFSET X=0 MOVE=1 MOVE_SPEED=300              # Clear X offset
	
		{% if printer.extruder.temperature < printer.extruder.target|float*0.85 %}
			M190 S{printer.extruder.target|float*0.85} # wait till 0.85 of bed temp is reached, then continue  
		{% endif %}
		SET_GCODE_OFFSET Z=-1 MOVE=1 MOVE_SPEED=400             # Adjust z-height
    
	{% endif %}

[gcode_macro T1]
gcode:
    {% if not printer.toolhead.extruder == "extruder1" %}
    ACTIVATE_EXTRUDER EXTRUDER=extruder1
    SET_GCODE_OFFSET Z=1 MOVE=1 MOVE_SPEED=400       # Adjust z-height
    SET_GCODE_OFFSET X=-20 MOVE=1 MOVE_SPEED=300     # Account for different X offset
    
	
	{% if printer.extruder1.temperature < printer.extruder1.target|float*0.85 %}
		M190 S{printer.extruder1.target|float*0.85} 		# wait till 0.85 of bed temp is reached, then continue  
	{% endif %}
	SET_GCODE_OFFSET Z=-1 MOVE=1 MOVE_SPEED=400      	# Adjust z-height
    
	{% endif %}
```

KevinOConnor, 2020-04-08:
I'm going to close this as it looks like the original question was answered (use g-code to raise the head prior to starting the wait for temperature).

-Kevin

closed 2020-04-08 by KevinOConnor