Klipper3d/klipper #2644: Weird layer shifts on X axis

iH8teF1ames, 2020-03-24:
Been having weird "layer shifts" ever since I installed klipper on my ender 3 (skr mini e3 tmc 2209) they arent like normal layer shifts however as instead of for the rest of the print it being shifted over by however much it initially shifted it moves back to where it is supposed to be without it re-homing any axis. I know this is not Z binding as it only appears on the X axis with the effects of it being seen on the Y axis. I have a very similar config on my CR-10 which exhibits none of these issues (same board, same drivers but Raspberry pi 3B+ instead of the 4 I have on my Ender 3) 
[image]
[image]
[klippy.log](https://github.com/KevinOConnor/klipper/files/4376901/klippy.log)
[image]

.

ld3300, 2020-03-25:
In your log it says that something in the klipper source changed.  Try updating to a clean install:
`git pull origin master`
`git reset --hard origin/master`

You also have a weird part of the log where there are tons of "M205" commands, which aren't supported by klipper, and shouldn't be repeating throughout a print anyway.  Either something weird happened in the slicer, or Klipper was repeating the message.

iH8teF1ames, 2020-03-25:
Trying that now. The new bed mesh sections had me messed up for a few minutes but I got that back up and running. As for the M205 commands, its a known issue with octoprint which doesn't really affect anything. I am currently running a test cube and we will see how that turns out.

iH8teF1ames, 2020-03-25:
Nope. Still showing the same layer shifts as before.

iH8teF1ames, 2020-03-25:
This is almost 100% a klipper related issue. I just flashed marlin on it and it did not show the issue at all.

KevinOConnor, 2020-03-29:
Have you tuned the tmc2209 current?  Best guess is that your X stepper has insufficient current.

Otherwise, if you already haven't, I'd recommend upgrading to the latest version of Klipper and retrying with that.

-Kevin

Kotletowsky, 2020-04-03:
I have similar issues. It appears only when new feature start in that layer. And only with "square" shapes. I'm using tmc2208
[image]

KevinOConnor, 2020-04-15:
Any further updates on this?

-Kevin

MagicRB, 2020-04-18:
I'm experiencing those layer shifts too, I don't know if they happen on the x axis, but I'll check. Though I'm sure none of my extruders skip a step, according to klipper stepper current is 0.8A. Also while I'm here, does it matter to which voltage are the little potentiometers set if I use my 2208 in UART mode?

MagicRB, 2020-04-18:
https://cdn.discordapp.com/attachments/272796072957444096/701051733480177784/IMG_20200418_144851.jpg
https://cdn.discordapp.com/attachments/272796072957444096/701051734176301096/IMG_20200418_144839.jpg
https://cdn.discordapp.com/attachments/272796072957444096/701051734952509460/IMG_20200418_144829.jpg
Some more examples, the one with worse overhangs was printed just now, and aside from the random z wobble of the one lead screw on my ender 3, they both share the distict layer shift at the top and at the bottom. The bottom one happens when the floor is finished, while the top one starts after of before the floor for the Z letter.

ld3300, 2020-04-21:
Check this out and see if it is related:
[https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/124](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/124)

MagicRB, 2020-04-21:
My printer does not produce a clunk, nor are the layer shifts that big, more like 0.1 mm

KevinOConnor, 2020-05-11:
I'm going to close this as it looks like the conversation has concluded.  To the best of my knowledge the Klipper code works as intended.

-Kevin

closed 2020-05-11 by KevinOConnor