Klipper3d/klipper #2648: Changing heater_bed max_power value through "button"

Murilolana, 2020-03-27:
Hi,

I'm trying to write a script to set the maximum bed heater power depending on outlet voltage. I live in Brazil and both 110v and 220v are used depending on the city. So, I have a 110v silicon heated bed and wish to set the max_power to 0.25 if connected to 220v line or 1.0 if connected to 110v line. I'll be using a 220v relay to change a pin state on the board. Theoretically, this setup would work as kind of "auto voltage selector" for the heat bed.
i've been trying to write something, but had no success.
I thought on something like:
{% if My_PIN == 0 %}
            { printer.heater_bed.max_power = 1 }
{% else %}
            { printer.heater_bed.max_power = 0.25 }
{% endif %}

Can anyone give a hand on this?

Thanks

[klippy.zip](https://github.com/KevinOConnor/klipper/files/4396002/klippy.zip)

ld3300, 2020-03-27:
Hi @Murilolana,
right now the max_power parameter is only set an initialization and isn't changeable at runtime.  The way I might accomplish it is to intercept the Bed G-code commands and modify them:

```
[gcode_button MY_PIN]
pin:
    <pin num>

[gcode_macros M140]
variable_power: 0
rename_existing: M1140
gcode:
    {power} = {S}
    {% if not QUERY_BUTTON button=MY_PIN %}
        {power} = {S * 0.25}
    M1140 S{POWER}

[gcode_macros M190]
variable_power: 0
rename_existing: M1190
gcode:
    {power} = {S}
    {% if not QUERY_BUTTON button=MY_PIN %}
        {power} = {S * 0.25}
    M1190 S{POWER}
```

Murilolana, 2020-03-30:
> Hi @Murilolana,
> right now the max_power parameter is only set an initialization and isn't changeable at runtime. You could set the max power to the safest value and it will just take longer to get there for the lower voltage. The Bed temperature sensor should start to PWM the output in either case as the heater gets hotter.
> 
> edit: removed bad answer

Was the piece of code bad? I was about to try it. Is it possible to intercept the M140/M190 and change it on the fly?

Thanks

ld3300, 2020-03-30:
Yes, you can intercept any Gcode since a few weeks ago.  The problem was I realized my solution would have changed the temperature of the bed, not the power delivery to get it there.  I basically had the wrong answer to the problem.  I imagine you want the bed temp to stay the same regardless of voltage.

Murilolana, 2020-03-30:
Sure, the bed temp should stay the same. I need just to adjust the maximum PWM according to the line voltage in order to match the same nominal power. It would be nice to have the possibility of changing htis parameter conditionally. I will try with both voltages to see if it I can a safe paremeter for both. I'm affraid that setting it to 0.3 and connecting to a 110v outlet wouldn't give the required power for getting the bed at 120ºC.

ld3300, 2020-03-30:
A hardware way you could do this is to switch a relay that runs the 220 through a diode so you are only getting half of the sine wave which would approximate the power of the 110 line.  You would just need a diode that can handle the wattage.

closed 2020-04-24 by KevinOConnor