Klipper3d/klipper #2650: Support for 20*04 LCD (HD44780) via I2C Connection

hraley, 2020-03-28:
Hi folks!

As per the title, I have tried in multiple ways to implement the connection of an I2C LCD display to my MCU (AVR Board - Atmega1280).

_I've recently returned to 3D printing after a hiatus of nearly 4 years. I have 4 machines all running the same MCU and LCD set up (with encoder) - although they all have different mechanical set ups.... Deltas, Cartesian, Core XY...etc...etc. 
Upgrading both legacy Marlin and Repetier Firmware to Klipper hasn't been too much of a task after I blew the cobwebs off my Github skills!
Awesome firmware, a true revolution, I'm damn impressed. I can bring my printers back into current spec for the price of a Raspberry Pi 3 A+ or a Pi Zero W.
I am using DWC2 as the web interface/control instead of Octoprint, after reading how streamlined it was in a cpu consumption sense!_

To run the LCD's previously, it was dead easy, Just connect the wires up to the SDA and SCL pins on the MCU, add the LiquidCrystalLibrary to Arduino IDE. enable in config! Job done!

> (This is the library I previously used - https://bitbucket.org/fmalpartida/new-liquidcrystal/src/integration/ )

Klipper already has support for the HD44780 (2004 LCD), just in a direct connected state. Could I do something to change the communication to I2C? From trawling through Gihub and other config files, i believe there is another type of display that uses I2C communication, so... it's there!!??

In the `[display]` section, I have tried defining `lcd_type: hd44780` with `spi_bus:` and `spi_speed:`.... to no avail.

Pictures....
[image]
Pictures.....
[image]

I hope someone might be able to point me in the correct direction, or at least educate me to why and how this does or doesn't work.
Hope I've provided enough info.

Kind Regards
Harry
[klippy.log](https://github.com/KevinOConnor/klipper/files/4397165/klippy.log)

KevinOConnor, 2020-03-29:
Unfortunately, Klipper does not currently have support for this.  Adding support would require an interested developer to implement and submit it.  (Unfortunately, it would likely require both host Python code and micro-controller C code.)

FWIW, you may be able to acquire an alternative display that Klipper does support - there are a number of low-cost displays available.

-Kevin

KevinOConnor, 2020-04-15:
I'm going to mark this as inactive as it doesn't look like a developer is actively working on it.

-Kevin

closed 2020-04-15 by KevinOConnor

urobasa, 2021-01-29:
Очень жаль, это хороший простой экран который я тоже использую с i2c

xsiriox, 2021-10-03:
Could be useful this to do it working? 

https://www.electroniclinic.com/raspberry-pi-16x2-lcd-i2c-interfacing-and-python-programming/