Klipper3d/klipper #2651: Z pos not changed after PROBE with different z_offset

pmarkiewicz, 2020-03-28:
This must be something stupid but I have no idea what.
I have delta printer with piezo probe and this works correctly.
What didn't work for me that regardless of z_offset after PROBE cmd z position is always same.
You can see in attached log first time z_offset is -0.3 next time is 1 and z is almost same after PROBE.
[klippy.log](https://github.com/KevinOConnor/klipper/files/4397410/klippy.log)

KevinOConnor, 2020-03-29:
It's not clear to me what is being reported.  It would help if you could describe the steps you took, the result you were expecting, and the result that you observed.

-Kevin

pmarkiewicz, 2020-03-29:
OK, let's start with first observed issue, after G1 Z100 position reported by  GET_POSITION is different (99.26) and this reported position is physically correct.

KevinOConnor, 2020-03-29:
That's certainly odd.  We'd need to see the Klipper log from that session in order to assist.  Run the steps that show the issue, run GET_POSITION, run M112, and then attach the resulting Klipper log file here.

-Kevin

pmarkiewicz, 2020-03-29:
Log attached, all I did is below
```
Send: G28
Recv: ok
[...]
Send: G1 Z200
Recv: ok
[...]
Send: GET_POSITION
Recv: // mcu: stepper_a:-3969 stepper_b:-3971 stepper_c:-4015
Recv: // stepper: stepper_a:388.674799 stepper_b:388.032575 stepper_c:388.703234
Recv: // kinematic: X:-0.645616 Y:0.408050 Z:199.263212
Recv: // toolhead: X:-0.645616 Y:0.408050 Z:199.264756 E:0.000000
Recv: // gcode: X:-0.645616 Y:0.408050 Z:200.000000 E:0.000000
Recv: // gcode base: X:0.000000 Y:0.000000 Z:0.000000 E:0.000000
[klippy.log](https://github.com/KevinOConnor/klipper/files/4399385/klippy.log)
[klippy.log](https://github.com/KevinOConnor/klipper/files/4399387/klippy.log)

Recv: // gcode homing: X:0.000000 Y:0.000000 Z:0.000000
Recv: ok
[...]
Send: M112
Send: N3 M112*34
Recv: // Klipper state: Shutdown

KevinOConnor, 2020-03-29:
You've enabled bed_tilt which alters the printer's understanding of the Z location.  For a description of GET_POSITION, see https://www.klipper3d.org/Code_Overview.html#coordinate-systems .  So, the printer knows you've commanded the Z to 100 (the "gcode: ... Z200") and knows the kinematic position is offset due to z tilt (hence "toolhead: ... Z199.26").

-Kevin

pmarkiewicz, 2020-03-29:
OK, this makes sense, so back to my original question.
I run probe with 2 different z_offset values but I don't see 1mm difference in z position after probe.
bed_tilt is disabled

```
#z_offset = -1.21
Send: probe
Recv: // probe at -0.646,0.408 is z=-0.346788
Recv: // Result is z=-0.346788
Recv: ok
[...]
Send: GET_POSITION
Recv: // mcu: stepper_a:-19898 stepper_b:-19768 stepper_c:-19901
Recv: // stepper: stepper_a:189.064799 stepper_b:188.422575 stepper_c:189.093234
Recv: // kinematic: X:-0.645616 Y:0.408050 Z:-0.346788
Recv: // toolhead: X:-0.645616 Y:0.408050 Z:-0.346788 E:0.000000
Recv: // gcode: X:-0.645616 Y:0.408050 Z:-0.098416 E:0.000000
Recv: // gcode base: X:0.000000 Y:0.000000 Z:0.000000 E:0.000000
Recv: // gcode homing: X:0.000000 Y:0.000000 Z:0.000000
Recv: ok
[...]
Send: RESTART
Recv: ok
Recv: // Klipper state: Disconnect
[...]
Recv: // Klipper state: Ready
[...]
Send: G28
Recv: ok
Send: G1 Z10 F2000
Recv: ok
#z_offset = -0.21
Send: PROBE
Recv: // probe at -0.646,0.408 is z=-0.336788
Recv: // Result is z=-0.336788
Recv: ok
Send: G1 Z5
Recv: ok
[...]
Send: probe
Recv: // probe at -0.646,0.408 is z=-0.296788
Recv: // Result is z=-0.296788
Recv: ok
Send: GET_POSITION
Recv: // mcu: stepper_a:-19895 stepper_b:-19763 stepper_c:-19896
Recv: // stepper: stepper_a:189.114799 stepper_b:188.472575 stepper_c:189.143234
Recv: // kinematic: X:-0.645616 Y:0.408050 Z:-0.296788
Recv: // toolhead: X:-0.645616 Y:0.408050 Z:-0.296788 E:0.000000
Recv: // gcode: X:-0.645616 Y:0.408050 Z:-0.048416 E:0.000000
Recv: // gcode base: X:0.000000 Y:0.000000 Z:0.000000 E:0.000000
Recv: // gcode homing: X:0.000000 Y:0.000000 Z:0.000000
Recv: ok

```

[klippy.log](https://github.com/KevinOConnor/klipper/files/4399885/klippy.log)

KevinOConnor, 2020-03-30:
It would help if you can explicitly state what it is you did, what you expected to see, and what you observed.

The PROBE command reports the Z height (as determined by the coordinate system defined by the stepper_a/b/c position_endstop config) when the probe triggers.  The probe's z_offset is not used during the PROBE command as it isn't relevant to that command.  (For reference, the probe's z_offset is used by various tools, such as DELTA_CALIBRATE.)

-Kevin

pmarkiewicz, 2020-03-30:
I do PROBE_CALIBRATE, z_offset is saved
Then I do restart, home, probe and G1 Z0.3 and this is 0.2mm too high. (I use 
gap gauge)
Next I change z_offset 1mm less and result is same, extruder is 0.2mm to high.
I repeat same doing delta_calibrate after homing and still no difference.

I just want to set correct z distance so I must miss something but have no idea what.

KevinOConnor, 2020-04-08:
I use the "endstop phase" feature to increase the accuracy of the endstops on my delta: https://www.klipper3d.org/Endstop_Phase.html

Otherwise, there currently isn't a good way to use the probe as a replacement for endstop accuracy.

-Kevin

pmarkiewicz, 2020-04-21:
OK, thank you.

closed 2020-04-21 by pmarkiewicz