Klipper3d/klipper #2654: MKS Robin Nano & TMC2208 UART problem

thucar, 2020-03-29:
I decided to switch my TMC2208's from stock standalone to UART.
I've only used SPI before, so this is a first.

I've soldered a single wire to the drivers UART pin and connected to a free IO pin on the Robin Nano. Then added TMC2208 config sections:

```
[stepper_x]
step_pin: PE3
dir_pin: !PE2
enable_pin: !PE4
step_distance: .0125
endstop_pin: !PA11
position_endstop: 0
position_max:230
homing_speed: 150

[tmc2208 stepper_x]
uart_pin:PC8
microsteps:16
interpolate: True
run_current:0.8
hold_current: 0.5
stealthchop_threshold: 300

[stepper_y]
step_pin: PE0
dir_pin: !PB9
enable_pin: !PE1
step_distance: .0125
endstop_pin: !PA12
position_endstop: 230
position_max:230
homing_speed: 150

[tmc2208 stepper_y]
uart_pin:PD12
microsteps:16
interpolate: True
run_current:0.8
hold_current: 0.5
stealthchop_threshold: 300
```
Unfortunately after booting the machine and restarting Klipper, I'm seeing an error: `Recv: !! Unable to read tmc uart 'stepper_x' register IFCNT`

The drivers are MKS version 2.0's with R8's in place. I can confirm that pins UART and PDN/UART are electrically connected. https://github.com/makerbase-mks/MKS-TMC2208/blob/master/MKS%20TMC2208%20V2.0_001/MKS%20TMC2208%20V2.0_001%20Layout.png

I've soldered the wire to PDN/UART

What else could I be missing?
The pins I use on the Robin Nano can be anything, correct? Since we're using software bit banging?

[sapphire.log](https://github.com/KevinOConnor/klipper/files/4398906/sapphire.log)

thucar, 2020-03-29:
Hrm, I just found a video by Makerbase, showing how to connect their TMC2209 drivers to their different boards. And they explicitly mention that on the Robin Nano board, only standalone (step/dir) mode can be used. UART mode is not available.

So it's not all in the software then? Some hardware conditions need to be met as well in order to use UART?

KevinOConnor, 2020-03-29:
The Klipper tmc uart code will work with any gpio pin on the stm32f103 chip.  However, if the board itself is grounding (or powering) the pdn-uart pin then it will interfere with the communication.  Also, if the board is supplying 5V io power to the step-stick (instead of 3.3V), then it would not be valid to wire pdn-uart directly to the mcu.

-Kevin

thucar, 2020-03-30:
Thank you @KevinOConnor. You pointed me in the right direction. One of the three jumpers located below stepper drivers also pulls the UART pin to VCC. Removing that jumper solved my issue.

closed 2020-03-30 by thucar