Klipper3d/klipper #2657: Stock creality Cr10 display is blank

bischofga55913, 2020-03-30:
I connected the display to my Mks-gen l 1.0v.
Than added this code to my printer.cfg:
`[display]
lcd_type: st7920

cs_pin: ar27

sclk_pin: ar25

sid_pin: ar29

encoder_pins: ^ar23, ^ar17

click_pin: ^!ar35`

I have restart everything.  The display doesnt show anything but blue.
The log for this attamp under the name klippy1.log.

Than i've tried the code from the example:

`########################################
# EXP1 / EXP2 (display) pins
########################################
#
[board_pins]
aliases:
    # EXP1 header
    EXP1_1=P1.31, EXP1_3=P0.18, EXP1_5=P0.15, EXP1_7=P1.0,  EXP1_9=<GND>,
    EXP1_2=P1.30, EXP1_4=P0.16, EXP1_6=P0.17, EXP1_8=P1.22, EXP1_10=<5V>,
    # EXP2 header
    EXP2_1=P0.8, EXP2_3=P3.25, EXP2_5=P3.26, EXP2_7=P0.27, EXP2_9=<GND>,
    EXP2_2=P0.7, EXP2_4=P0.28, EXP2_6=P0.9,  EXP2_8=<RST>, EXP2_10=<NC>
   # Pins EXP2_1, EXP2_6, EXP2_2 are also MISO, MOSI, SCK of bus "ssp1"

# See the sample-lcd.cfg file for definitions of common LCD displays.
[display]
lcd_type: st7920
cs_pin: EXP1_7
sclk_pin: EXP1_6
sid_pin: EXP1_8
encoder_pins: ^EXP1_5, ^EXP1_3
click_pin: ^!EXP1_2

[output_pin beeper]
pin: EXP1_1`

When I try this, octoprint reports:
 Gemeldeter Fehler: Unknown value 'P1.0' in enumeration 'pin'
The .log  for this attamp is: klippy2.log.

I hope someone can help me. 
[klippy1.log](https://github.com/KevinOConnor/klipper/files/4401585/klippy1.log)
[klippy2.log](https://github.com/KevinOConnor/klipper/files/4401586/klippy2.log)

Thanks in advance.

KevinOConnor, 2020-03-31:
If you have an MKS GEN-L board then you should use the "board_pins" snippet from config/generic-ramps.cfg along with a display snippet from config/sample-lcd.cfg.

-Kevin

bischofga55913, 2020-04-02:
Thanks for your answer. 
But this isn't working for me. I have followed your instructions, but the display is still empty. Here is the current klippy.log.
[klippy.log](https://github.com/KevinOConnor/klipper/files/4423325/klippy.log)

Thanks for your Answer
Gabriel Bischof

KevinOConnor, 2020-04-08:
Alas, that's the best information I have on that particular board.  Some of the boards / displays use a different orientation for the plug, which could be an issue for you.  If so, there are numerous tutorials online discussing how to detect and fix that.

-Kevin

closed 2020-04-24 by KevinOConnor