Klipper3d/klipper #2658: SET_LED INDEX not stable in printer.cfg

Sonsi79, 2020-03-30:
I can use the new gcode "SET_LED ... INDEX=1/2/3"
because the LCD ist splited in 3 LED Areas. 
By this way it is possible to set each area to different colors.

How to set this as initial default in printer.cfg?

KevinOConnor, 2020-04-02:
It is not currently possible to set different LEDs to different colors from the config file.  It is only possible via the SET_LED g-code command.  The closest equivalent would be to add it to a START_PRINT macro that is called at the start of a print, or alternatively to use a delayed_gcode config setting (see config/example-extras.cfg for the details).

-Kevin

ld3300, 2020-04-13:
@Sonsi79 add this to your config file:

```
# Things printer should do on connection (variables, etc).  Will only run once
[delayed_gcode ON_CONNECT]
initial_duration: 0.5      # Wait until 0.5 seconds after printer reports READY
gcode:
    #set led color here and any other startup things you want
```

Sonsi79, 2020-04-14:
> @Sonsi79 add this to your config file:
>
>
>
> ```
>
> # Things printer should do on connection (variables, etc).  Will only run once
>
> [delayed_gcode ON_CONNECT]
>
> initial_duration: 0.5      # Wait until 0.5 [...]

Thnx lot. Works great for me!

closed 2020-04-14 by Sonsi79