Klipper3d/klipper #2659: Print failing almost immediately.

dash3811, 2020-03-30:
I have installed Klipper and wen through the config checks and everything is working fine. however, when I try to do the test for pressure advance it will almost immediately stop. I have attached my log file as well as the gcode from simplify3d.

[square_tower.zip](https://github.com/KevinOConnor/klipper/files/4402414/square_tower.zip)

KevinOConnor, 2020-03-31:
The log does not indicate any errors occurred.  If you think this is a software issue, be sure to issue an M112 command after the undesirable event, and then attach the log from that session - see https://www.klipper3d.org/Contact.html .

-Kevin

dash3811, 2020-03-31:
I got it to work after disabling octolapse. Is it not compatible with that plugin?

RomRider, 2020-03-31:
Octolapse is working with klipper but it is probably misconfigured for your setup. Run octolapse in test mode and check the logs from octolapse to undestand what happens.
[image]

dash3811, 2020-04-01:
I am unsure why it keeps failing but I'm more concerned with why M106 is returning an unknown command response. 

[image]

closed 2020-04-03 by dash3811

ld3300, 2020-04-13:
It looks like it might do that if you don't give it a value ie `M106 S<value>`

dash3811, 2020-04-13:
I figured it out. my wiring was wonky