Klipper3d/klipper #2660: Possible to set TMC2209 stealthchop threshold mid print?

HWiese1980, 2020-03-30:
I'm currently printing something and have the stealthchop threshold set fairly low. Now the printer is constantly switching between SpreadCycle and StealthChop during accelerations which makes a lot of noise. Is there a way to change the threshold during a running print?

No logs needed for this case, I suppose...

KevinOConnor, 2020-03-31:
It's possible to alter the TMC driver settings via the SET_TMC_FIELD command, but there is no tool to set a new stealthchop threshold to a mm/s value at runtime.  In general though, I'd recommend setting the stealthchop threshold above the max velocity for the axis (for stealthchop always enabled), or very low (eg, less than 5mm/s).  Changing between stealthchop/spreadcycle at velocity is not something the tmc drivers handle well.

-Kevin

KevinOConnor, 2020-04-15:
I'm going to close this as it looks like the original question was answered (it is possible via SET_TMC_FIELD, but isn't particularly easy).

-Kevin

closed 2020-04-15 by KevinOConnor