Klipper3d/klipper #2662: Z offset ignord?

Coffee0297, 2020-03-30:
hello kevin. 

i have some issues atm, when i do a probe calibrate and save that, its waaaaay over the bed.
like 3-4mm. it have workd in past so dont know why it sudently ignores my offset. here is a log showing me making the offset and then homing and moving head to the bed. at the estop its 3-4mm over the bed. 
[klippy.log](https://github.com/KevinOConnor/klipper/files/4403201/klippy.log)
[image]

-Tonny

Coffee0297, 2020-03-30:
but if i put 7.55 in the DO_NOT_EDIT for the Bl Touch its fine. why do klipper say the offset is 4 when its not? 
-Tonny

ld3300, 2020-03-30:
did you run PROBE_CALIBRATE correctly?

Coffee0297, 2020-03-31:
yes, im quite familiar with all klipper commands :)

Coffee0297, 2020-04-01:
hello @KevinOConnor, 
here is a log more. 
[klippy.log](https://github.com/KevinOConnor/klipper/files/4414250/klippy.log)

**i have not touch the machine between tests. same with config**

steps ive done.
- home the machine, 
- do a  probe_calibrate to a piece of paper
- accept and save_config
- home again
- move the print head down with g1z1 and g1 z0.2. **here the nozzle are perfect height**
- when i start a print it digs the nozzle 3 or 4mm **inside** the bed... 

why do klipper all of a sudden not respect the offsets? 
i can set the offset 3-4mm higher and the g1 z0.2 is waaaaay off the bed, but when print starts its fine. 
why do this happen? and how can i help getting the bug away? im really dependent on my machine atm ( printing visors for hospitals).. and i don't wanna go marlin or duet if this can be solved.

Kind regards Tonny

Coffee0297, 2020-04-01:
did a GET_POSITION  and got a weird result... after a probe_calibrate, formware restart and home i got this when nozzle are at printing hight, eg 0.2mm over bed
[image]
kind regards Tonny

Coffee0297, 2020-04-01:
So i did some more debug and found a bug in the Z-tilt code
if you don't do z-tilt and probe its fine, correct offset and all. but if you do a ztilt after the probe calibrate it goes 3mm under the bed.

is there anyone that can spot the error? and make a fix? i can't do python...

KevinOConnor, 2020-04-01:
I don't see anything incorrect in the log.  It appears the configuration of the probe z_offset parameter is incorrect.  You have a bl-touch configured, but (under normal circumstances) the bl-touch can not operate correctly with a 4mm z_offset.  Double check the [directions for probe_calibrate](https://www.klipper3d.org/Probe_Calibrate.html) (eg, make sure the x_offset and y_offset are correct).  If you're still having trouble, attach the klipper log file from a probe_calibrate session.

-Kevin

Coffee0297, 2020-04-01:
ill update the title, its not the calibrate that seems to be the issue after debugging. its the z-tilt

- Home => Probe Calibrate => Home => Z Tilt = **broke**
- Home => Z Tilt => Probe Calibrate => Home => Z Tilt = **broke**
- Home => Z Tilt => Probe Calibrate => Home = **not broke**
- Home=> Probe Calibrate => home = **not broke**

kind regards Tonny

Coffee0297, 2020-04-01:
found a temp fix. 

If i home Z after a z-tilt its back to the right offset and such. 
i don't think that is the expected behavior? 

I've always done it like this and only in the most recent times it has become an issue.
- Home => Z Tilt => Print.

^^ had that running for like 10 months no issue

It has nothing to do with the probe_calibrate, it was just how i found out. 
so after the change that has been done somewhere i need to do.
- Home => Z Tilt  Home Z => Print.

kind regards Tonny

Coffee0297, 2020-04-03:
i have a other printer where it works as exspected. 
FIRMWARE_VERSION:v0.8.0-328-g4a0dcc6a FIRMWARE_NAME:Klipper
the printer where its buged is:
Firmware version: v0.8.0-395-g3c01b60a
Kind regards Tonny

sniperlucian, 2020-04-05:
just updated klipper to lates and can see the same - crushed my bed ...

KevinOConnor, 2020-04-08:
I'm not aware of any recent changes to z_tilt.  If a regression has occurred it would help to run through "git bisect" to determine which commit introduced the regression.

-Kevin

KevinOConnor, 2020-04-24:
Any further updates on this?

-Kevin

Coffee0297, 2020-04-24:
Im trying to get some prints done. So i cant test atm. But it's easy to recreate, so most shuld be able to find it

KevinOConnor, 2020-05-11:
I'm going to close this as it looks like the conversation has concluded.  To the best of my knowledge the Klipper code works as intended.

-Kevin

closed 2020-05-11 by KevinOConnor