Klipper3d/klipper #2663: X-Y Shifts During Print

ryanA3D, 2020-03-30:
I installed Klipper on my Anycubic Kossel Linear Plus a couple of days ago, bed level via basic and detailed delta calibrate, and have run a few other checks via config_checks.md and slicers.md.  All test prints thus far show odd layer shifts in X and Y though it appears to self correct during the print and then shift again.  See attached pics.  Occurs a few times in each XYZ Cube I've done.  Sliced in various gcode flavors marlin, smoothie, reprap, none of those seem to matter.  I've run prints at varying accelerations ranging from 1000, to 2000, to 2500, to 3000, doesn't make a difference.  I've also slowed down speeds greatly, that also has no effect.  I was running Marlin fw prior to upgrading and didnt have this issue, so I dont believe it to be a physical machine issue.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4405252/klippy.log)
[image]
[image]
[image]
[image]

KevinOConnor, 2020-04-08:
Typical reasons for this are heater instability (either at the bed or nozzle), insufficient current to the stepper motor drivers, and possibly too high a movement speed.  Otherwise, nothing stands out in the log as an obvious error.

-Kevin

ryanA3D, 2020-04-21:
I took everything apart, greased the rails, reassembled and tightened belts and it appears this issue is now gone.  Thanks for the help, closing.

closed 2020-04-21 by ryanA3D

micdzi85, 2020-06-19:
I have simmilar issue. I have installed klipper on raspberry pi 4 to run my Anycubic Kossel Linear plus. I have changed the stepper drivers to stepstick TMC2208 v2.0 (pic attached). They are well cooled by an external fan and I've set the V-ref to what was suggested. 
I'm getting a layer shift some time into the print. even despite running slow (40 mm/s).

[image]

[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/4804724/klippy.log.zip)

can you help?

micdzi85, 2020-06-19:
I forgot to mention - never had any issues with this setup running Marlin 1.9...Klipper lured me with a promis of speed an quality - both are there... where it actually prints between the shifted layers....

ryanA3D, 2020-06-19:
For my issue, the fix was a few things.  Tuning the stepper driver vref to a proper value, and some maintenance on my printer.

1. For the Vref setting, we need to know some electrical properties of our stepper motors, namely that they are nema 17 stepper motors and have a rated/peak current of 1.5A.  https://www.thingiverse.com/groups/anycubic-kossel/forums/general/topic:33486.   Vref == Imax == 1.5V (https://learn.watterott.com/silentstepstick/faq/).  I would recommend running Vref between 1.4 - 1.5V.  With your cooling solution, you shouldn't overheat.  If you are seeing high temps, you can scale this value down a bit, but as you're aware too low or too high current on the stepper drivers can result in skipped steps and artifacts or shifts in the prints.

2.  I also performed a good bit of maintenance on my printer.  Grease the rails, tighten the belt tension if they are starting to get slack in them, check the screw connections on the pulley heads to make sure they're tight, check the screws connecting the rods to make sure they're tight.  Also, if you're observing a lot of vibration in the frame when printing at higher speeds, you can slow down or stiffen the frame.  There are several frame stiffening options on thingiverse that can be printed and attached.

micdzi85, 2020-06-19:
Hi, 

thanks for the input. I have the frame stiffened with corner braces, lately printed the belt tensioners (https://www.thingiverse.com/thing:2827665) and upgraded to a belt with wire inlay - hoping for little stretch. I was considering changing the rods to magnetic, so I dismanteled the effector to check the size differences. they are big, but also the rods are damn expensie to ship to where I live, so I'll try to leave it for the last step, however I'll check if all is tight.
I'll have a look at the Vref - when installing it with Malin, I've set it to 0,9 on towers and 1,1 on extruder.  I'll at least try to match it to extruder's (there are no skips there) and then go up from there. 
thanks!