Klipper3d/klipper #2665: Probing does not take X/Y offsets into account

thefotios, 2020-03-31:
I know this is essentially a dupe of https://github.com/KevinOConnor/klipper/issues/1803 (as well as covering some points in other issues) but this seems to affect quite a few places like `safe_z_home`, `bed_mesh`, and `screws_tilt_adjust`. At the very least maybe the docs need some clarification. But I also think that if there was an extra flag like proposed in #1803, it would make configuration much simpler on users while also maintaining backwards compatibility.

This could also be a good place where a boot time config check similar to how `position_endstop` needs to be between `position_min` and `position_max` is validated.

I'd be happy to take a shot at implementing this if there is interest in it.

---
[klippy.log](https://github.com/KevinOConnor/klipper/files/4406032/klippy.log)

thefotios, 2020-04-04:
I also noticed another odd inconsistency. It appears the `SCREWS_TILT_CALCULATE` takes the offsets into account, but `BED_MESH_CALIBRATE` does not.

If set `screw1` to `10,35` it works correctly. But if I set `mesh_min` to `10,35` I get the following error:
```
Send: BED_MESH_CALIBRATE
Recv: !! Move out of range: -32.000 45.000 5.000 [0.000]
```

KevinOConnor, 2020-04-08:
This appears to be a dup of #2510.  

-Kevin

nicedevil007, 2020-04-08:
@thefotios @KevinOConnor 

while the #2510 is closed now I can't ask there for help with excatly the same problem.
The offset of my BL-Touch isn't taken in account while doing a `BED_MESH_CALIBRATE`

`SCREWS_TILT_CALCULATE` is working as it should be.

I did a calculation of my mesh_min and mesh_max that the measured points are all symetrical.
The error is absolutly the same just with other x/y values.

I also read through your guide and nothing changes to me after adjusting everything to the described possible values [https://github.com/KevinOConnor/klipper/blob/master/docs/Bed_Mesh.md]

Here you can find my config for an Artillery Sidewinder X1 with a printbed of 300x300

[https://pastebin.com/QT6TPD17]

ld3300, 2020-04-13:
The min and max values for BED_MESH_CALIBRATE have to be the relative values to your probe, not nozzle.  As Kevin was pointing out by referencing #2510 is that this is identified in the documentation. See the 20200109 post in [https://github.com/KevinOConnor/klipper/blob/master/docs/Config_Changes.md#changes](https://github.com/KevinOConnor/klipper/blob/master/docs/Config_Changes.md#changes)

KevinOConnor, 2020-04-24:
I'm going to close this in favor of the existing #2510 .

-Kevin

closed 2020-04-24 by KevinOConnor