Klipper3d/klipper #2667: Printer.cfg file for SOVOL SV01

lalek54, 2020-03-31:
Hi. 
I made a printer.cfg file for the SOVOL Sv01 printer and I d like to share it. 
How can I do?
Here is my file. 
[gcode_macro START_PRINT]
default_parameter_BED_TEMP: 60
default_parameter_EXTRUDER_TEMP: 190
gcode:
    #Reset Feedrate and flow rate
    M220 S100 
    M221 S100 
    # Use absolute coordinates
    G90
    # Reset the G-Code Z offset (adjust Z offset if needed)
    SET_GCODE_OFFSET Z=-0.100
    #Reset Extruder
    G92 E0
    #Move Z Axis up
    G1 Z2.0 F3000

[gcode_macro M900]
gcode: SET_PRESSURE_ADVANCE ADVANCE={K}

[gcode_macro PARK_MACRO]
default_parameter_X: 0
default_parameter_Y: 0
default_parameter_Z: 30
gcode:
        M117 PARKING..
        G91
        G1 E-3.14 F1500
        G1 Z{Z}
	G90
	G1 X{X} Y{Y} F3000
	
[gcode_macro UNLOAD]
gcode:
	G91
        G1 E5.0 F1200
        G1 E3.0 F1600
        G1 E-13.14 F7000
	G1 E-540 F3000
	G90
        
[gcode_macro M600]
gcode:
	PAUSE_MACRO
	UNLOAD

[gcode_macro PAUSE_MACRO]
gcode:
	PAUSE
	PARK_MACRO
	
[gcode_macro PURGE]
gcode:
        M117 PURGING..
        G91
        G1 E45.0 F250
        G90
        
[gcode_macro LOAD_FILAMENT]
gcode:
        M117  LOADING...
        G91
        G1 E25.0 F1000
        G1 E435 F2500
        G4 P900
        G1 E45.0 F250
        G90

[gcode_macro RESUME_MACRO]
gcode:
        M117  RESUMING...
        RESUME
        OCTO10   

[stepper_x]
step_pin: ar54
dir_pin: ar55
enable_pin: !ar38
step_distance: .0125
endstop_pin: ^ar3
position_endstop: 0
position_max: 300
homing_speed: 50

[stepper_y]
step_pin: ar60
dir_pin: ar61
enable_pin: !ar56
step_distance: .0125
endstop_pin: ^ar14
position_endstop: 0
position_max: 255
homing_speed: 50

[stepper_z]
step_pin: ar46
dir_pin: !ar48
enable_pin: !ar62
step_distance: .0025
endstop_pin: ^ar18
position_endstop: 0
position_max: 300

[extruder]
pressure_advance: 0.05
#   The amount of raw filament to push into the extruder during
#   extruder acceleration. An equal amount of filament is retracted
#   during deceleration. It is measured in millimeters per
#   millimeter/second. The default is 0, which disables pressure
#   advance.
#pressure_advance_smooth_time: 0.01
#   A time range (in seconds) to use when calculating the average
#   extruder velocity for pressure advance. A larger value results in
#   smoother extruder movements. This parameter may not exceed 200ms.
#   This setting only applies if pressure_advance is non-zero. The
#   default is 0.040 (40 milliseconds).
step_pin: ar26
dir_pin: !ar28
enable_pin: !ar24
step_distance: .0024
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: ar10
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog13
control: pid
pid_Kp: 31.147
pid_Ki: 2.076
pid_Kd: 116.803
min_temp: 0
max_temp: 265

[heater_bed]
heater_pin: ar8
sensor_type: ATC Semitec 104GT-2
sensor_pin: analog14
control: pid
pid_Kp: 72.174
pid_Ki: 1.816
pid_Kd: 717.224
min_temp: 0
max_temp: 110

[filament_switch_sensor my_sensor]
switch_pin: ar2
pause_on_runout: True

[fan]
pin: ar9

[mcu]
serial: /dev/ttyUSB0
pin_map: arduino

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 6000
max_z_velocity: 5
max_z_accel: 100
acceleration_order: 6
min_jerk_limit_time: 0.03  

[display]
lcd_type: st7920
cs_pin: ar16
sclk_pin: ar23
sid_pin: ar17
encoder_pins: ^ar33, ^ar31
click_pin: ^!ar35

[virtual_sdcard]
path: ~/.octoprint/uploads/

[pause_resume]
recover_velocity: 50.

[firmware_retraction]
#retract_length: 1
#   The length of filament (in mm) to retract when G10 is activated, and to
#   unretract when G11 is activated (but see unretract_extra_length below).
#   The default is 0 mm.
#retract_speed:50
#   The speed of retraction, in mm/s. The default is 20 mm/s.
#unretract_extra_length: 0
#   The length (in mm) of *additional* filament to add when unretracting.
unretract_speed: 50
#   The speed of unretraction, in mm/s. The default is 10 mm/s.

lalek54, 2020-04-01:
Can you help me to reference my file as your example of printer.cfg please. 
It will be useful as this printer is becoming more and more popular. 
Thanks.

Zulaner, 2020-04-28:
Thanks, worked well

KevinOConnor, 2020-04-30:
Thanks.  I added a basic config (commit 1460e0d6).

-Kevin

closed 2020-05-11 by KevinOConnor

Iceman231173, 2020-06-12:
Funktioniert diese CFG auch für das SKR mini V1.2 ? oder nur für das Originale Board ?