Klipper3d/klipper #2668: Z-axis step-loss due to hard braking on probe_triggered (possibly?)

Rudgas, 2020-03-31:
Setup:

CoreXY Hypercube evolution
Dual Z rods
Dual drivers (TMC2130)

Stepper settings:
```
[stepper_z]
step_pin: P0.22
dir_pin: !P2.11
enable_pin: !P0.21
step_distance: 0.00015625 #0.000625 #0.0025
endstop_pin: probe: z_virtual_endstop
position_endstop: 0.5
position_min: -5
position_max: 250

[stepper_z1]
step_pin: P0.1
dir_pin: !P0.0
enable_pin: !P0.10
step_distance: 0.00015625 #0.000625 #0.0025

[printer]
kinematics: corexy
max_velocity: 200
max_accel: 2000
max_z_velocity: 5
max_z_accel: 3
```

TMC settings:
```
[tmc2130 stepper_z]
cs_pin: P1.10
spi_software_miso_pin: P0.5
spi_software_mosi_pin: P4.28
spi_software_sclk_pin: P0.4
##diag1_pin: P1.25
microsteps: 256
run_current: 1.000
hold_current: 0.500
stealthchop_threshold: 100
interpolate: False

[tmc2130 stepper_z1]
cs_pin: P1.1
spi_software_miso_pin: P0.5
spi_software_mosi_pin: P4.28
spi_software_sclk_pin: P0.4
##diag1_pin: P
microsteps: 256
run_current: 1.000
hold_current: 0.500
stealthchop_threshold: 100
interpolate: False
```

Testing procedure:

PROBE_ACCURACY PROBE_SPEED=5 SAMPLES=150
PROBE_ACCURACY at X:108.500 Y:73.900 Z:2.000 (samples=150 retract=1.000 speed=5.0 lift_speed=5.0)

Result:
```
Recv: // probe at 108.500,73.900 is z=0.795937
Recv: // probe at 108.500,73.900 is z=0.799687
Recv: // probe at 108.500,73.900 is z=0.799531
Recv: // probe at 108.500,73.900 is z=0.799375
Recv: // probe at 108.500,73.900 is z=0.799531
Recv: // probe at 108.500,73.900 is z=0.799531
Recv: // probe at 108.500,73.900 is z=0.799531
Recv: // probe at 108.500,73.900 is z=0.799687
Recv: // probe at 108.500,73.900 is z=0.799687
Recv: // probe at 108.500,73.900 is z=0.799531
Recv: // probe at 108.500,73.900 is z=0.799375
Recv: // probe at 108.500,73.900 is z=0.799531
Recv: // probe at 108.500,73.900 is z=0.799531
Recv: // probe at 108.500,73.900 is z=0.799531
Recv: // probe at 108.500,73.900 is z=0.799219
Recv: // probe at 108.500,73.900 is z=0.799531
Recv: // probe at 108.500,73.900 is z=0.799531
Recv: // probe at 108.500,73.900 is z=0.799687
Recv: // probe at 108.500,73.900 is z=0.799687
Recv: // probe at 108.500,73.900 is z=0.799375
Recv: // probe at 108.500,73.900 is z=0.799531
Recv: // probe at 108.500,73.900 is z=0.799687
Recv: // probe at 108.500,73.900 is z=0.799375
Recv: // probe at 108.500,73.900 is z=0.799531
Recv: // probe at 108.500,73.900 is z=0.799219
Recv: // probe at 108.500,73.900 is z=0.799375
Recv: // probe at 108.500,73.900 is z=0.799375
Recv: // probe at 108.500,73.900 is z=0.799531
Recv: // probe at 108.500,73.900 is z=0.799375
Recv: // probe at 108.500,73.900 is z=0.799375
Recv: // probe at 108.500,73.900 is z=0.799531
Recv: // probe at 108.500,73.900 is z=0.799219
Recv: // probe at 108.500,73.900 is z=0.799531
Recv: // probe at 108.500,73.900 is z=0.799531
Recv: // probe at 108.500,73.900 is z=0.799687
Recv: // probe at 108.500,73.900 is z=0.799531
Recv: // probe at 108.500,73.900 is z=0.799687
Recv: // probe at 108.500,73.900 is z=0.799844
Recv: // probe at 108.500,73.900 is z=0.799844
Recv: // probe at 108.500,73.900 is z=0.799844
Recv: // probe at 108.500,73.900 is z=0.801094
Recv: // probe at 108.500,73.900 is z=0.800312
Recv: // probe at 108.500,73.900 is z=0.800000
Recv: // probe at 108.500,73.900 is z=0.800000
Recv: // probe at 108.500,73.900 is z=0.800156
Recv: // probe at 108.500,73.900 is z=0.799375
Recv: // probe at 108.500,73.900 is z=0.800469
Recv: // probe at 108.500,73.900 is z=0.800469
Recv: // probe at 108.500,73.900 is z=0.800625
Recv: // probe at 108.500,73.900 is z=0.800625
Recv: // probe at 108.500,73.900 is z=0.800625
Recv: // probe at 108.500,73.900 is z=0.800781
Recv: // probe at 108.500,73.900 is z=0.800781
Recv: // probe at 108.500,73.900 is z=0.800937
Recv: // probe at 108.500,73.900 is z=0.800937
Recv: // probe at 108.500,73.900 is z=0.800937
Recv: // probe at 108.500,73.900 is z=0.801094
Recv: // probe at 108.500,73.900 is z=0.800937
Recv: // probe at 108.500,73.900 is z=0.801094
Recv: // probe at 108.500,73.900 is z=0.801094
Recv: // probe at 108.500,73.900 is z=0.801406
Recv: // probe at 108.500,73.900 is z=0.801406
Recv: // probe at 108.500,73.900 is z=0.801562
Recv: // probe at 108.500,73.900 is z=0.801406
Recv: // probe at 108.500,73.900 is z=0.801719
Recv: // probe at 108.500,73.900 is z=0.801719
Recv: // probe at 108.500,73.900 is z=0.801875
Recv: // probe at 108.500,73.900 is z=0.802031
Recv: // probe at 108.500,73.900 is z=0.802187
Recv: // probe at 108.500,73.900 is z=0.802031
Recv: // probe at 108.500,73.900 is z=0.802031
Recv: // probe at 108.500,73.900 is z=0.802344
Recv: // probe at 108.500,73.900 is z=0.802187
Recv: // probe at 108.500,73.900 is z=0.802187
Recv: // probe at 108.500,73.900 is z=0.802187
Recv: // probe at 108.500,73.900 is z=0.802187
Recv: // probe at 108.500,73.900 is z=0.802812
Recv: // probe at 108.500,73.900 is z=0.802969
Recv: // probe at 108.500,73.900 is z=0.802656
Recv: // probe at 108.500,73.900 is z=0.802969
Recv: // probe at 108.500,73.900 is z=0.803125
Recv: // probe at 108.500,73.900 is z=0.803125
Recv: // probe at 108.500,73.900 is z=0.803125
Recv: // probe at 108.500,73.900 is z=0.803281
Recv: // probe at 108.500,73.900 is z=0.803437
Recv: // probe at 108.500,73.900 is z=0.803594
Recv: // probe at 108.500,73.900 is z=0.803594
Recv: // probe at 108.500,73.900 is z=0.803750
Recv: // probe at 108.500,73.900 is z=0.803906
Recv: // probe at 108.500,73.900 is z=0.803906
Recv: // probe at 108.500,73.900 is z=0.803594
Recv: // probe at 108.500,73.900 is z=0.803281
Recv: // probe at 108.500,73.900 is z=0.803750
Recv: // probe at 108.500,73.900 is z=0.803906
Recv: // probe at 108.500,73.900 is z=0.803750
Recv: // probe at 108.500,73.900 is z=0.804062
Recv: // probe at 108.500,73.900 is z=0.804219
Recv: // probe at 108.500,73.900 is z=0.804531
Recv: // probe at 108.500,73.900 is z=0.805156
Recv: // probe at 108.500,73.900 is z=0.804219
Recv: // probe at 108.500,73.900 is z=0.804219
Recv: // probe at 108.500,73.900 is z=0.804687
Recv: // probe at 108.500,73.900 is z=0.804375
Recv: // probe at 108.500,73.900 is z=0.804531
Recv: // probe at 108.500,73.900 is z=0.805000
Recv: // probe at 108.500,73.900 is z=0.805156
Recv: // probe at 108.500,73.900 is z=0.805469
Recv: // probe at 108.500,73.900 is z=0.805469
Recv: // probe at 108.500,73.900 is z=0.805469
Recv: // probe at 108.500,73.900 is z=0.805469
Recv: // probe at 108.500,73.900 is z=0.805469
Recv: // probe at 108.500,73.900 is z=0.805469
Recv: // probe at 108.500,73.900 is z=0.805625
Recv: // probe at 108.500,73.900 is z=0.805625
Recv: // probe at 108.500,73.900 is z=0.805781
Recv: // probe at 108.500,73.900 is z=0.805625
Recv: // probe at 108.500,73.900 is z=0.805469
Recv: // probe at 108.500,73.900 is z=0.804844
Recv: // probe at 108.500,73.900 is z=0.805000
Recv: // probe at 108.500,73.900 is z=0.804375
Recv: // probe at 108.500,73.900 is z=0.804844
Recv: // probe at 108.500,73.900 is z=0.805469
Recv: // probe at 108.500,73.900 is z=0.804844
Recv: // probe at 108.500,73.900 is z=0.805312
Recv: // probe at 108.500,73.900 is z=0.806875
Recv: // probe at 108.500,73.900 is z=0.806406
Recv: // probe at 108.500,73.900 is z=0.806875
Recv: // probe at 108.500,73.900 is z=0.806875
Recv: // probe at 108.500,73.900 is z=0.806094
Recv: // probe at 108.500,73.900 is z=0.806719
Recv: // probe at 108.500,73.900 is z=0.807344
Recv: // probe at 108.500,73.900 is z=0.806875
Recv: // probe at 108.500,73.900 is z=0.807031
Recv: // probe at 108.500,73.900 is z=0.807031
Recv: // probe at 108.500,73.900 is z=0.806875
Recv: // probe at 108.500,73.900 is z=0.806875
Recv: // probe at 108.500,73.900 is z=0.807031
Recv: // probe at 108.500,73.900 is z=0.806875
Recv: // probe at 108.500,73.900 is z=0.804844
Recv: // probe at 108.500,73.900 is z=0.805781
Recv: // probe at 108.500,73.900 is z=0.805937
Recv: // probe at 108.500,73.900 is z=0.806719
Recv: // probe at 108.500,73.900 is z=0.806719
Recv: // probe at 108.500,73.900 is z=0.806562
Recv: // probe at 108.500,73.900 is z=0.805781
Recv: // probe at 108.500,73.900 is z=0.806094
Recv: // probe at 108.500,73.900 is z=0.806719
Recv: // probe at 108.500,73.900 is z=0.807031
Recv: // probe at 108.500,73.900 is z=0.807031
Recv: // probe at 108.500,73.900 is z=0.806094
Recv: // probe accuracy results: maximum 0.807344, minimum 0.795937, range 0.011406, average 0.802679, median 0.802266, standard deviation 0.002729
```

So I observed a small but continuous drift in the Z-distance, indicating that the Bed is moving further and further away on each sample, this drift is minimal ~0.01mm over 150 samples but nonetheless very reproducible at different speeds and accelerations. So I'm wondering if maybe the abrupt deceleration when the Z-probe is triggered might be the cause of this?

Rudgas, 2020-04-01:
Or easier, can I set a maximum deceleration when the probe triggers?

FanDjango, 2020-04-01:
No comment, but see #2509 and welcome to the club

Rudgas, 2020-04-01:
Can be closed as seems to be same issue as #2509

closed 2020-04-01 by Rudgas