Klipper3d/klipper #2670: ILI9328 support

thucar, 2020-03-31:
I'm looking into getting the display working on my Sapphire Pro. So far I've managed to deduct that the screen connected to the Robin Nano board is based on ILI9328. Managed to find the datasheet for it and found out that I need to pull pins high/low to configure it to communicate over SPI.

Will run some real life tests tomorrow, started this issue to keep track of progress and hopefully get pointers. 

The first step is to get it working as a display. The perfect world end goal would be to get touch working as well.

[ILI9328.pdf](https://github.com/KevinOConnor/klipper/files/4410025/ILI9328.pdf)

thucar, 2020-03-31:
@KevinOConnor I'm seeing references to FSMC in the lib/stm32f1/system_stm32f1xx.c file. Also it seems the display is connected to Robin Nano via a FSMC connector. 

Is there a relation between the two? Meaning, would it make sense to look into adding support for FSMC displays in general to Klipper?

KevinOConnor, 2020-04-15:
FWIW, I don't have any knowledge of that display, so I don't have any meaningful guidance.  I'm sure it's possible to add support; a developer would need to find out all the details and implement it.

-Kevin

thucar, 2020-04-16:
I’ll try and claw my way through this at some point. I see that this and many other displays are directly supported by the U8Glib that Marlin uses. Maybe that would be the way to go - to find and include a display library that handles support for multiple popular displays so we won’t have to reinvent the wheel.

KevinOConnor, 2020-05-11:
Okay - I'm going to add the "inactive" label - just to group it with other feature requests that aren't currently in a "blocking" state.

-Kevin

closed 2020-05-11 by KevinOConnor